zephyrproject-rtos/zephyr #5980: NRF5 I2C standard speed 250kHz

thomafred, 2018-02-05:
The NRF5 I2C-driver SCL-line measures 250kHz when configured with standard speed (`I2C_SPEED_STANDARD`), while fast speed (`I2C_SPEED_FAST`) measures 100kHz.

The i2c-bus specifiation lists the following bidirectional speeds (section 5 of [this document](https://www.nxp.com/docs/en/user-guide/UM10204.pdf))

* **Standard-mode (Sm)** - bit rates up to 100 kbit/s
* **Fast-mode (Fm)** - bit rates up to 400 kbit/s
* **Fast-mode Plus (Fm+)** - bit rate up to 1 Mbit/s
* **High-speed mode (Hs-mode)** - bit rate up to 3.4 Mbit/s

anangl, 2018-02-07:
250 kHz is the default frequency set in the TWI peripheral after reset. It looks like the configuration register is not set at all. Are you sure that the `i2c_configure` function is successful (i.e. it returns 0)?
Perhaps you don't use the macro `I2C_SPEED_SET` when configuring the speed?
When I do:
```
int result = i2c_configure(i2c_dev, I2C_SPEED_SET(I2C_SPEED_STANDARD));
```
the speed is configured correctly.

carlescufi, 2018-09-04:
Not a bug.

closed 2018-09-04 by carlescufi