zephyrproject-rtos/zephyr #5982: nRF5x  subscribe will cause HardFault while disconnect and reconnect

sam131208, 2018-02-05:
While  discovery the Current time Service and subscribe to server(iphone is server), and then disconnect from the setting page on iphone.  and open the bluetooth on setting page again, reconnect the nRF5x device. the zephyr will run to  the  function  att_handle_rsp(),  and the  function pointer att->req->func is NULL, while run the func(att->chan.chan.conn, err, pdu, len, att->req) will cause a HardFault. 

   The  function  att_handle_rsp() is list below:

static u8_t att_handle_rsp(struct bt_att *att, void *pdu, u16_t len, u8_t err){
	bt_att_func_t func;

	BT_DBG("err %u len %u: %s", err, len, bt_hex(pdu, len));

	/* Cancel timeout if ongoing */
	k_delayed_work_cancel(&att->timeout_work);

	if (!att->req) {
		BT_WARN("No pending ATT request");
		goto process;
	}

	/* Release original buffer */
	if (att->req->buf) {
		net_buf_unref(att->req->buf);
		att->req->buf = NULL;
	}

	/* Reset func so it can be reused by the callback */
	func = att->req->func;
	att->req->func = NULL;

	func(att->chan.chan.conn, err, pdu, len, att->req);

	/* Don't destroy if callback had reused the request */
	if (!att->req->func) {
		att_req_destroy(att->req);
	}

	att->req = NULL;

process:
	/* Process pending requests */
	att_process(att);

	return 0;
})

Vudentz, 2018-02-08:
Can we have a trace with CONFIG_BT_ATT_DEBUG and CONFIG_BT_GATT_DEBUG? Note that the API of bt_gatt_subscribe do state that struct bt_gatt_subscribe_params *params should not be touched during its operation, perhaps there is something cleaning up the _req used by the stack to track the reply?

nashif, 2018-02-13:
@sam131208 ping, can you please provide the logs?

sam131208, 2018-03-02:
@nashif  @Vudentz ,under is the log and the source code.

[subscribe.log](https://github.com/zephyrproject-rtos/zephyr/files/1773476/subscribe.log)
[test.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/1773479/test.tar.gz)

PR #6482 (2018-03-06): Bluetooth: GATT: Fix possible crash while resubscribing
When reconnecting the code will attempt to recover the subscriptions
but it was not setting any callback causing the bt_att_req.func to be
NULL.

Fixes #5982
commit 9c92609a (2018-03-06): Bluetooth: GATT: Fix possible crash while resubscribing

When reconnecting the code will attempt to recover the subscriptions
but it was not setting any callback causing the bt_att_req.func to be
NULL.

Fixes #5982
closed 2018-03-06 by carlescufi