zephyrproject-rtos/zephyr #5994: samples/bluetooth/ipsp: build failed for MICRO-BIT & NRF51-PCA10028 HW

tarunkum, 2018-02-06:
I was trying to build the ipsp test from samples/bluetooth folder for the target MICRO-BIT & NRF51-PCA10028 HW.

Getting some build issue if I am trying to build for these respective targets. It seems SRAM space issue occurs.

Below is the error message appears on terminal.

[1/148] Generating always_rebuild
Building for board nrf51_pca10028
[144/148] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf 
: && ccache sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -T zephyr/linker.cmd -Wl,-Map=/home/zephyr/samples/bluetooth/ipsp/build/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -e__start -Wl,--start-group -Wl,--whole-archive libapp.a zephyr/libzephyr.a zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a zephyr/subsys/bluetooth/controller/libsubsys__bluetooth__controller.a zephyr/subsys/net/libsubsys__net.a zephyr/subsys/net/ip/libsubsys__net__ip.a zephyr/subsys/net/ip/l2/libsubsys__net__ip__l2.a zephyr/subsys/net/lib/app/libsubsys__net__lib__app.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -Wl,--end-group -L"/home/sysroots/armv5-zephyr-eabi/usr/lib/arm-zephyr-eabi/6.2.0/armv6-m" -L/home/zephyr/samples/bluetooth/ipsp/build/zephyr -lgcc -Wl,--print-memory-usage -nostartfiles -nodefaultlibs -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--gc-sections -Wl,--build-id=none && :
Memory region         Used Size  Region Size  %age Used
           FLASH:      149004 B       256 KB     56.84%
            SRAM:       39736 B        32 KB    121.26%
        IDT_LIST:         132 B         2 KB      6.45

sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: zephyr/zephyr_prebuilt.elf section `noinit' will not fit in region `SRAM'
sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: section .intList VMA [0000000020008000,0000000020008083] overlaps section noinit VMA [00000000200023e0,0000000020008a67]
sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: region `SRAM' overflowed by 6968 bytes
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.

Vudentz, 2018-02-08:
It simply doesn't fit:
SRAM: 39736 B 32 KB 121.26%

There maybe possible to fine-tune the buffers sizes to make them fit on 32 KB, but given it is almost 7 KB over I think it will be pretty hard to squeeze.

nashif, 2018-02-11:
closing, sample does not fit, reduce configs or the feature is just too big for the board.

closed 2018-02-11 by nashif