zephyrproject-rtos/zephyr #5996: Need a "ps aux" like command to list all running threads and their attributes

jli157, 2018-02-06:
Can Zephyr have a "ps aux" like command, to list all of current running threads, with their detailed information, like 
1. Is it a user mode thread of kernel thread? 
2. How much is its stack size? 
3. Thread Id number. 
4. ..

This command should be very helpful for debugging purpose.

nashif, 2018-02-06:
the shell actually has some basic version of that, try samples/subsys/shell/shell_module/ and run 

> kernel threads

```
Zephyr Shell, Zephyr version: 1.10.99
Type 'help' for a list of available commands
shell> kernel threads
Threads:
*0x004011e0:   options: 0x0 priority: -9
 0x004014b4:   options: 0x0 priority: -1
 0x00401280:   options: 0x1 priority: 15
shell>
```

I had a more detailed version with more data, but I lost it in some branch somewhere, need to search for it again :(

jli157, 2018-02-06:
That is great! I think this command can meet my requirement. But if you have a better solution, that should be better. :) 

Thank you!

jli157, 2018-02-08:
Well documented. So, it can be closed.

closed 2018-02-08 by jli157