zephyrproject-rtos/zephyr #6015: PWM on 32bit arch can get 0 pulse_cycle because of 64bit calculation

tmauron, 2018-02-07:
In the file zephyr/include/pwm.h, 64 bit is used to calculate the period an the pulse. On a 32 bit machine we might get `0` for the `pulse_cycles` or `period_cycles`. A suggestion could be to change `pulse_cycles = (pulse * cycles_per_sec) / USEC_PER_SEC;` to `pulse_cycles = (cycles_per_sec / USEC_PER_SEC) * pulse;`
Here is a snippet of the modified function that works for me:

```
static inline int pwm_pin_set_usec(struct device *dev, u32_t pwm,
				   u32_t period, u32_t pulse)
{
	u32_t period_cycles, pulse_cycles, cycles_per_sec;

	if (pwm_get_cycles_per_sec(dev, pwm, (u32_t*)&cycles_per_sec) != 0) {
		return -EIO;
	}
	
	period_cycles = (cycles_per_sec / USEC_PER_SEC) * period;
	pulse_cycles = (cycles_per_sec / USEC_PER_SEC) * pulse;

	return pwm_pin_set_cycles(dev, pwm, period_cycles, pulse_cycles);
}
```

nashif, 2018-02-20:
can you submit a PR?

carlescufi, 2018-12-06:
@gamnes @pizi-nordic could you please take a look at this one to see if it's still applicable?

nashif, 2020-03-18:
no movement for 2 years, closing

closed 2020-03-18 by nashif