zephyrproject-rtos/zephyr #6017: Enable easy use of shields with overlays

erwango, 2018-02-07:
Most of boards available in Zephyr are compatible with various shields of different kind, most common on being Arduino type.
There is a wide range of these shields and providing a easy way to use them would help faster prototyping and demos. Thanks to device tree and overlays there is only a little to do to make this available.

I propose to add a SHIELD= option in build command. Shield .overlay file would be merged with board dts files before processing. 
Shield overlays should define node names that would be proposed aliases by the boards. Files could be stored under boards/shield or be picked up in external app/ build environment.

For instance:
boards/shields/x-nucleo-iks01a1.overlay
`&arduino_i2c{
    hts221@5f {
        compatible = "st,hts221";
        reg = <0x5f>;
        label = "HTS221";
    };

    lps25hb-press@5d {
        compatible = "st,lps25hb-press";
        reg = <0x5d>;
        label = "LPS25HB";
    };

    lis3mdl-magn@1e {
        compatible = "st,lis3mdl-magn";
        reg = <0x1e>;
        label = "LIS3MDL";
    };

    lsm6ds0@6b {
        compatible = "st,lsm6ds0";
        reg = <0x6b>;
        label = "LSM6DS0";
    };
};`

On a Arduino compatible board, only modification would be
'    aliases{
        arduino_i2c = &i2c1;
    };'

PS: Having issue to enable code snippets. If someone manages to make it better, don't hesitate to edit

carlescufi, 2018-02-07:
@erwango fixed your snippets, you need to use 3 backticks (`) without the quotes:
"```"
<code>
"```"

mbolivar, 2018-02-07:
Adding @superna9999 since he sent email about this to the list a while back

SebastianBoe, 2018-02-08:
This sounds great :)

"Thanks to device tree and overlays there is only a little to do to make this available."

I would be very interested to understand how device tree makes this easier. E.g., is there an ecosystem of shield device tree descriptions that we can leverage? Also, a description of what the attached device tree snippets are describing and how that information integrates with the rest of the build system would be great.

SebastianBoe, 2018-02-08:
"I propose to add a SHIELD= option in build command."

I would propose that this option be a Kconfig option.

erwango, 2018-02-09:
> E.g., is there an ecosystem of shield device tree descriptions that we can leverage? 

Not that I am aware of

> Also, a description of what the attached device tree snippets are describing and how that information integrates with the rest of the build system would be great.

Here are the detailed operations generated for one sensors, based on the example above.
I hope this will answer your question (please  tell me otherwise).

```
&arduino_i2c{
    lis3mdl-magn@1e {
        compatible = "st,lis3mdl-magn";
        reg = <0x1e>;
        label = "LIS3MDL";
    };
 };
```
This help generating the following defines:
```
#define CONFIG_LIS3MDL_DEV_NAME	HTS221
#define CONFIG_LIS3MDL_I2C_ADDR		0x1E
```
And, since this sensor is defined in arduino_i2c node. If this alias is defined in board.dts:
```
    aliases{
        arduino_i2c = &i2c1;
    };
```
 with 
```
		i2c1: i2c@40005400 {
			compatible = "st,stm32-i2c-v1";
			clock-frequency = <I2C_BITRATE_STANDARD>;
			#address-cells = <1>;
			#size-cells = <0>;
			reg = <0x40005400 0x400>;
			interrupts = <31 0>, <32 0>;
			interrupt-names = "event", "error";
			status = "disabled";
			label= "I2C_1";
		};
```
then I2C is enabled, configured and sensor is associated with its bus master label
```
#define CONFIG_LIS3MDL_I2C_MASTER_DEV_NAME	I2C_1
```

> I would propose that this option be a Kconfig option.

I would stand with my initial cmake option proposal. Once available, it would allow to avoid touching Kconfig files or when one wants to propose a new shields.overlay.
Besides, I think it makes sense, as there is a sort of logic to specify the shield in the command where you specify the board.

SebastianBoe, 2018-02-12:
> I would stand with my initial cmake option proposal. Once available, it would allow to avoid touching Kconfig files or when one wants to propose a new shields.overlay.
> Besides, I think it makes sense, as there is a sort of logic to specify the shield in the command where you specify the board.

True, it does make sense to specify daughterboards (shields) in the same way as the motherboard (BOARD) is specified. But, using CMake variables to specify the board is not a good idea either,
rather it is something we must do for legacy and technical reasons.

About avoiding touching Kconfig files. It is not clear to me how one developer would share his BOARD+Shield configuration with another developer when using CMake variables. Perhaps some run.sh script that called CMake with the appropriate -DSHIELD=foo command?

With Kconfig however, there exist mechanisms for documenting the option, discovering the option, sharing a configuration instance of the option etc. So if it is technically possible to use Kconfig I would recommend so over CMake variables.

erwango, 2018-02-12:
> With Kconfig however, there exist mechanisms for documenting the option, discovering the option, sharing a configuration instance of the option etc..

These are valid points, I'll have a try with Kconfig as soon as I have some time.

mbolivar, 2018-02-12:
DTC_OVERLAY_FILE supports multiple files now. If you are using one overlay file for each "shield", you don't need a separate SHIELD variable. Just add the overlays to the DTC_OVERLAY_FILE list, separated by spaces.

erwango, 2018-03-13:
@mbolivar : Just seen your comment. Thanks for the input.

I have been convinced by argument from @sebo, saying that Kconfig will help storing configuration into git and share with others (while a command line is not made to be stores).

I'll use your input as a solution to realize this.

erwango, 2018-03-15:
Hi @SebastianBoe ,

I just had a try: https://github.com/erwango/zephyr/tree/shield_overlay).
Plan was to define CONFIG_SHIELD_X_NUCLEO_IKS01A1, then based on this, do some Kconfig computation and add boards/shields/x_nucleo_iks01a1/x_nucleo_iks01a1.overlay to DTC_OVERLAY_FILE.

Problem is that dts computation happens _before_  Kconfig.
Hence, I'm able to add .overlay (only one for now) to DTC_OVERLAY_FILE, but it's too late.

Anyway idea how I could squeeze Kconfig stuff before dts?
(It should be doable as ARCH is required for dts.cmake. Or maybe this is not the same as config ARCH...)

carlescufi, 2018-03-15:
@erwango I am not sure I understand:

>Problem is that dts computation happens before Kconfig.

That is not the case, Kconfig is processed first, and then DTS is run:

Kconfig:
https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/app/boilerplate.cmake#L241

DTS:
https://github.com/zephyrproject-rtos/zephyr/blob/242c583c9652bd961b1c96f409eb915cc5c9a024/cmake/app/boilerplate.cmake#L260

>Anyway idea how I could squeeze Kconfig stuff before dts?

We do use Kconfig in DTS itself, so there's something I'm missing here:

https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/dts.cmake#L18

erwango, 2018-03-16:
@carlescufi : you're right, I've been mixing Kconfig and CMake processing.

So point is I'm assigning DTC_OVERLAY_FILE in boards/shield/CMakeLists.txt and it happens to be done after dts compialtion happens (cf https://pastebin.com/LkZHSNCz, look for "EGO").
I hope this is more accurate.

erwango, 2018-03-16:
Anyway, this helped me. Will try to check how to put that in dts.cmake, rather than in boards/CMakeLists.txt that happen to be computed too late.
Thanks!

erwango, 2018-03-16:
So, it's working now (https://github.com/erwango/zephyr/tree/shield_overlay),
but with several limitations:
A]. "aliases" not working as expected. "arduino_i2c", as described above, could not be used as a node name, this is not supported by dtc.
B]. Matching dts.fixup requires to be provided for each board
C]. only one shield taken into account

A] is a blocker. Support of this feature requires a way to bind shield nodes with varying nodes. In the example above board ardunio I2C connector could be i2c1, i2c2, i2ca, ... It should be possible to bind
shield i2c to any board i2c, without knowing it a priori.
So I've by-passed this limitation by using "i2c1" as node name for shield overlay. But, of course,
this doesn't scale for use with any board.

B] will prevent practically to use the feature as fixup definitions should be provided for all boards.
This really lower the interest of the feature. But eventually, .fixup files will disappear...

C] Given points A] and B], I haven't looked further to this point. I assume there should be a way to do it,
even if it could be tricky or requires to modify a bit zephyr build.

To conclude, feature is blocked by missing way to bind shields nodes into boards (A]).
Then, while we still cary dts.fixup files, there is only a little interest for this feature (Main interest for now would be to document and provide configuration to use shields).

I'll try to look for alternate solution for A], but with lower prio (at least untill B] get solved).

SebastianBoe, 2018-03-20:
Is it possible to solve B) by having an entry in common.dts?

#ifdef CONFIG_SHIELD
#include "shield.overlay"
#endif

See how we solved a similair, but easier, problem in https://github.com/zephyrproject-rtos/zephyr/commit/72684ef815ef594af54ed446b21730480f20641f#diff-e9f991f72df8e2fe46241e1236069780

superna9999, 2018-03-20:
It's a way to solve, but it won't solve the multiple shields issue

erwango, 2018-03-28:
Thanks @SebastianBoe, I'll have a look.

Meanwhile, working with @galak, we've come up to several things that should help to get things working.

A] Could be overcome by defining a connector node outside soc
```arduino,i2c: &i2c1{};```
This require dtc v1.4.2 or later and some modification in devicetree.py
(Commit of interest: https://git.kernel.org/pub/scm/utils/dtc/dtc.git/commit/?h=v1.4.6&id=3346e065aacb1f05bdfb3872e3ba709b5f895846)

B] Could be overcome by consequence of A] or by use of device tree aliases.
Aim is to be able to generate:
```#define ARDUINO_I2C_ST_HTS221_5F_BUS_NAME	ST_STM32_I2C_V1_40005400_ST_HTS221_5F_BUS_NAME```
```#define ARDUINO_I2C_ST_HTS221_5F_LABEL		ST_STM32_I2C_V1_40005400_ST_HTS221_5F_LABEL```
```#define ARDUINO_I2C_ST_HTS221_5F_BASE_ADDRESS	ST_STM32_I2C_V1_40005400_ST_HTS221_5F_BASE_ADDRESS```
Then in boards/shield/*/dts.fixup:
```#define CONFIG_HTS221_NAME			ARDUINO_I2C_ST_HTS221_5F_LABEL```
```#define CONFIG_HTS221_I2C_MASTER_DEV_NAME	ARDUINO_I2C_ST_HTS221_5F_BUS_NAME```

C] Should be possible by hacking in extensions.cmake

I'm pushing a pull request to make things visible and ease discussions

erwango, 2018-11-22:
Merged some time ago, closing

closed 2018-11-22 by erwango