zephyrproject-rtos/zephyr #6023: Invalid behaviour of Transport Layer after Incomplete timer expiration

MariuszSkamra, 2018-02-07:
Instead of ignore segments of cancelled message, Host send acknowledgment message.
According to Mesh v1.0 3.5.3.4 Reassembly behavior:
"When the incomplete timer expires, the lower transport layer shall consider that the message being
received has failed and cancel the acknowledgment timer. Any segment of a canceled message shall be ignored."
In this test case PTS is not sending 8th segment even if we are sending ACK, waiting to incomplete timer to expire. After expiration of this timer PTS resends segments 2, 6 (appartently didn't received our ACK) and 8th. We shall ignore those. Instead of that ACK is sent

Looking into the logs:
```
[bt] [WRN] seg_ack: Incomplete timer expired
[bt] [DBG] send_ack: (0x20002aa0) SeqZero 0x0006 Block 0x00000000 OBO 0
[bt] [DBG] bt_mesh_ctl_send: (0x20002aa0) src 0x0002 dst 0x0001 ttl 0xff ctl 0x00
[bt] [DBG] bt_mesh_ctl_send: (0x20002aa0) len 6: 001800000000
[bt] [DBG] bt_mesh_net_send: (0x20002aa0) src 0x0002 dst 0x0001 len 7 headroom 9 tailroom 13
[bt] [DBG] bt_mesh_net_send: (0x20002aa0) Payload len 7: 00001800000000
[bt] [DBG] bt_mesh_net_send: (0x20002aa0) Seq 0x000054
[bt] [DBG] bt_mesh_net_encode: (0x20002aa0) src 0x0002 dst 0x0001 ctl 1 seq 0x000054
[bt] [DBG] seg_rx_reset: (0x20002aa0) rx 0x20007358
[bt] [DBG] bt_mesh_net_recv: (0x20001944) rssi -25 net_if 0
[bt] [DBG] bt_mesh_net_decode: (0x20001944) 29 bytes: 7055fe1dd342aabc85ea8e1ea325ea66427f441c9a6ab48bf82af78525
[bt] [DBG] net_find_and_decrypt: (0x20001944) 
[bt] [DBG] friend_decrypt: (0x20001944) NID 0x70 net_idx 0x0000
[bt] [DBG] net_decrypt: (0x20001944) NID 0x70 net_idx 0x0000
[bt] [DBG] net_decrypt: (0x20001944) IVI 0 net->iv_index 0x0000000a
[bt] [DBG] net_decrypt: (0x20001944) src 0x0001
[bt] [DBG] bt_mesh_net_decode: (0x20001944) Decryption successful. Payload len 25
[bt] [DBG] bt_mesh_net_decode: (0x20001944) src 0x0001 dst 0x0002 ttl 3
[bt] [DBG] bt_mesh_net_decode: (0x20001944) PDU: 700300000f00010002e88018c8313f753a7f9b952c46120455
[bt] [DBG] bt_mesh_trans_recv: (0x20001944) src 0x0001 dst 0x0002 seq 0x0000000f friend_match 0
[bt] [DBG] bt_mesh_trans_recv: (0x20001944) Payload e88018c8313f753a7f9b952c46120455
[bttester] [DBG] net_recv_ev: ttl 0x03 ctl 0x00 src 0x0001 dst 0x0002 payload_len 16
[bt] [DBG] trans_seg: (0x20001944) ASZMIC 1 AKF 1 AID 0x28
[bt] [DBG] trans_seg: (0x20001944) SeqZero 0x0006 SegO 6 SegN 8
[bt] [DBG] seg_rx_alloc: (0x20001944) New RX context. Block Complete 0x000001ff
[bt] [DBG] trans_seg: (0x20001944) Received 6/8
[bt] [DBG] bt_mesh_net_recv: (0x20001944) rssi -23 net_if 0
[bt] [DBG] bt_mesh_net_decode: (0x20001944) 29 bytes: 702bd6c7c07d5009acaea86cae3c6d421d22a687a1599f1b0af565bdf0
[bt] [DBG] net_find_and_decrypt: (0x20001944) 
[bt] [DBG] friend_decrypt: (0x20001944) NID 0x70 net_idx 0x0000
[bt] [DBG] net_decrypt: (0x20001944) NID 0x70 net_idx 0x0000
[bt] [DBG] net_decrypt: (0x20001944) IVI 0 net->iv_index 0x0000000a
[bt] [DBG] net_decrypt: (0x20001944) src 0x0001
[bt] [DBG] bt_mesh_net_decode: (0x20001944) Decryption successful. Payload len 25
[bt] [DBG] bt_mesh_net_decode: (0x20001944) src 0x0001 dst 0x0002 ttl 3
[bt] [DBG] bt_mesh_net_decode: (0x20001944) PDU: 700300001000010002e8801908a94936fc832f65efe4833e4d
[bt] [DBG] bt_mesh_trans_recv: (0x20001944) src 0x0001 dst 0x0002 seq 0x00000010 friend_match 0
[bt] [DBG] bt_mesh_trans_recv: (0x20001944) Payload e8801908a94936fc832f65efe4833e4d
[bttester] [DBG] net_recv_ev: ttl 0x03 ctl 0x00 src 0x0001 dst 0x0002 payload_len 16
[bt] [DBG] trans_seg: (0x20001944) ASZMIC 1 AKF 1 AID 0x28
[bt] [DBG] trans_seg: (0x20001944) SeqZero 0x0006 SegO 8 SegN 8
[bt] [DBG] trans_seg: (0x20001944) Existing RX context. Block 0x00000040
[bt] [DBG] trans_seg: (0x20001944) Target len 8 * 12 + 12 = 108
[bt] [DBG] trans_seg: (0x20001944) Received 8/8
[bt] [DBG] seg_ack: (0x20002aa0) rx 0x20007358
[bt] [DBG] send_ack: (0x20002aa0) SeqZero 0x0006 Block 0x00000140 OBO 0
```

Looking into the code of seg_ack (transport.c):
```
	if (k_uptime_get_32() - rx->last > K_SECONDS(60)) {
		BT_WARN("Incomplete timer expired");
		send_ack(rx->sub, rx->dst, rx->src, rx->ttl,
			 &rx->seq_auth, 0, rx->obo);
		seg_rx_reset(rx, true);
```
this send_ack call looks suspicious to me

Apart from that, in trans_seg (transport.c):
````
	/* Look for old RX sessions */
	rx = seg_rx_find(net_rx, seq_auth);
	if (rx) {
(....)
		/* We ignore instead of sending block ack 0 since the
		 * ack timer is always smaller than the incomplete
		 * timer, i.e. the sender is misbehaving.
		 */
		BT_WARN("Got segment for canceled SDU");
		return -EINVAL;
	}
````
Seems that we don't hit this, new rx context is created instead
```
[bt] [DBG] seg_rx_alloc: (0x20001944) New RX context. Block Complete 0x000001ff
```

I enclose debug logs from test case execution.
Affected TC: MESH/NODE/TNPT/BV-12-C

MariuszSkamra, 2018-02-07:
debug logs [iut-zephyr.log](https://github.com/zephyrproject-rtos/zephyr/files/1703189/iut-zephyr.log)
test execution logs [autoptsclient-zephyr.log](https://github.com/zephyrproject-rtos/zephyr/files/1703187/autoptsclient-zephyr.log)

jhedberg, 2018-02-13:
@MariuszSkamra what's the latest on this one? Did you do a full test run the patch that you showed on IRC a few days ago?

MariuszSkamra, 2018-02-14:
I will run the full test-run now. I ran Network layer and Transport layer tests and it seemed to not introduce regressions. When I'm done, I'll put feedback here.

MariuszSkamra, 2018-02-14:
@jhedberg Patch does not introduce regression

PR #6207 (2018-02-14): Bluetooth: Mesh: Ignore segments of cancelled message
This fixes the issue when after incomplete timer expiration
host sent ACK.
The host failed in two cases:
1. Sending ACK right after the incomplete timer expiration;
2. Sending ACK from new RX context. Now, seq_auth of cancelled
message is not cleaned on RX reset, so segments of cancelled message
will be discarded when resend.

According to the Mesh Profile v1.0
"When the incomplete timer expires, the lower transport layer
shall consider that the message being received has failed and
cancel the acknowledgment timer. Any segment of a canceled [... cut]
commit daa8040e (2018-02-14): Bluetooth: Mesh: Ignore segments of cancelled message

This fixes the issue when after incomplete timer expiration
host sent ACK.
The host failed in two cases:
1. Sending ACK right after the incomplete timer expiration;
2. Sending ACK from new RX context. Now, seq_auth of cancelled
message is not cleaned on RX reset, so segments of cancelled message
will be discarded when resend.

According to the Mesh Profile v1.0
"When the incomplete timer expires, the lower transport layer
shall consider that the message being received has failed and
cancel the acknowledgment timer. Any segment of a cancele [... cut]
closed 2018-02-14 by jhedberg