zephyrproject-rtos/zephyr #6026: subsys: storage: flash_map - power management feature

nvlsianpu, 2018-02-07:
flash_map module provide flash_area_open and flash_area_close API. Is it possibile that original (so MyNewt) intention for these API call was to provide some kind ow power management regards memory devices.

This discusion started at [convert FCB to use sectors PR](https://github.com/zephyrproject-rtos/zephyr/pull/5987)

<b>flash power management future:</b>

@ccollins476ad wrote:

The changes look good to me. I just have one possible concern. I noticed the flash area gets opened once in fcb_init() (via a call to flash_area_open()), and never gets closed. For internal flash, this is not a problem, but I believe the expectation was that flash areas would be opened when accessed, and closed when the access is complete.
I guess there was a thought that keeping an external flash device open is problematic (perhaps for power usage reasons).
So I'm not sure if this is something that should be changed or not. Like I said, it should work fine with internal flash.
Maybe it is best to keep your changes as is. This seems like an issue that is best addressed when needed (if ever).

my answer to this:
so far calling flash_area_open each time only increasing the latency. flash_area_close dose nothing. Maybe it was an idea of power management basing on this, but so fare this add only unwanted latency. My proposition is to modify flash_map API instead: Add f. flash_area_get(int flash_area_id, flash_area **fap) - it will do what current flash_area_open' does so it will return proper flash_area pointer basing on flash_area_id. Change 'int flash_area_open(int flash_area_id, flash_area **fap)' to 'int flash_area_open(flash_area *fap)' and limit its functionality to "flash power activation features" - so it will be empty for now. We will keepflash_area_close` as is - but we coment that it is dedicated to functionality of "flash power deactivation future".

@mbolivar - I need your opinion on this, I'm not sure what idea was behind flash_area_open/close API i mynewt.

^^
@OpenSourceFoundries

nvlsianpu, 2018-02-07:
cc @ccollins476ad @mbolivar

mbolivar, 2018-02-07:
@nvlsianpu I have often wondered the same thing.

In the middle of last year, I inserted some debugging calls into mcuboot's loader.c file to track opening and closing, and it was full of bugs. There were a lot of times that things were getting opened but never closed, and the newt implementations for those functions were empty IIRC.

I fixed the mcuboot bugs as best I could (with few exceptions, we now have that boot_go() basically just opens everything in the beginning and closes it in the end), and then left it there.

Maybe the newt people can shed more light on this -- though as I've said before I don't know how much sense it makes to duplicate newt internal APIs in Zephyr. It seems like a "shortest path" approach, but I don't like it in general.

de-nordic, 2024-02-23:
This has already been discussed here: https://github.com/zephyrproject-rtos/zephyr/pull/60709.
TL;DR:  open/close functions are supposed only to get ownership and access to given flash area only and special power management functions should be provided instead.
Subsystems open areas and never release them, which is intended, and this would make device work at elevated power level.