zephyrproject-rtos/zephyr #6038: Update Zephyr Licensing page

carlescufi, 2018-02-07:
This is missing several of the components in `ext/`
http://docs.zephyrproject.org/LICENSING.html#zephyr-licensing

dbkinder, 2018-02-14:
TSC approved [Contributing guidelines](http://docs.zephyrproject.org/contribute/contribute_non-apache.html) say all the ext/ components need a README file with specific origin and licensing information. None of them do.  There's a template for this README in ``doc/contribute/code_component_README``.

carlescufi, 2018-03-06:
@Shilpashree3lc will this be done in time for the scheduled Friday release?

Shilpashree3lc, 2018-03-07:
Here is my finding on the external components which are in compliant and not in compliant with the template as of the README in ``doc/contribute/code_component_README``. 

ext component | Path | Comments
-- | -- | --
st | ext/hal/st/stm32cube/stm32f1xx/README | Not in Compliant
st | ext/hal/st/stm32cube/stm32f4xx/README | Not in Compliant
st | ext/hal/st/stm32cube/stm32f7xx/README | Not in Compliant
st | ext/hal/st/stm32cube/stm32f0xx/README | Not in Compliant
st | ext/hal/st/stm32cube/stm32f3xx/README | Not in Compliant
st | ext/hal/st/stm32cube/README | Not in Compliant
st | ext/hal/st/stm32cube/stm32l4xx/README | Not in Compliant
st | ext/hal/st/lib/sensor/vl53l0x/README | Compliant
st | ext/hal/st/lib/README | Not in compliant
ti | ext/hal/ti/simplelink/README | Not in compliant
cmsis | ext/hal/cmsis/README | Not in compliant
silabs | ext/hal/silabs/gecko/README | Not in compliant
qmsi | ext/hal/qmsi/README | Not in compliant
nordic | ext/hal/nordic/nrfx/README | compliant
altera | ext/hal/altera/README | compliant
nxp | ext/hal/nxp/mcux/README | Not in compliant
hal | ext/hal/README | Not in compliant
fat | ext/fs/fat/README | Not in compliant
fs | ext/fs/README | Not in compliant
tinycbor | ext/lib/encoding/tinycbor/README | Not in compliant
tinycrypt | ext/lib/crypto/tinycrypt/README | Not in Compliant
mbedtls | ext/lib/crypto/mbedtls/README | Not in Compliant
mcumgr | ext/lib/mgmt/mcumgr/README.md | Not in compliant
nffs | ext/fs/nffs/README.md | Not in compliant
atmel | ext/atmel | No README
esp | ext/esp | No README

Shilpashree3lc, 2018-03-07:
To Start with, I am listing the current status and [Plan]

1.	Only below #3 are in compliant state – complete in all sense
a.	ext/hal/st/lib/sensor/vl53l0x/README
b.	ext/hal/nordic/nrfx/README
c.	ext/hal/altera/README

2.	Below #2 have all together NO information – **[Plan]** Suggest to contact the respective external owners here
a.	ext/atmel
b.	ext/esp

3.	Below #21 do have information but NO fully compliant as we expect – **[Plan]** I can start to work on them, not possible to finish by wk10.5
a.	ext/hal/st/stm32cube/stm32f3xx/README
b.	ext/hal/st/stm32cube/README
c.	ext/hal/st/stm32cube/stm32l4xx/README
d.	ext/hal/st/lib/README
e.	ext/hal/ti/simplelink/README
f.	ext/hal/cmsis/README
g.	ext/hal/silabs/gecko/README
h.	ext/hal/qmsi/README
i.	ext/hal/nxp/mcux/README
j.	ext/hal/README
k.	ext/fs/fat/README
l.	ext/fs/README
m.	ext/lib/encoding/tinycbor/README
n.	ext/lib/crypto/tinycrypt/README
o.	ext/lib/crypto/mbedtls/README
p.	ext/lib/mgmt/mcumgr/README.md
q.	ext/fs/nffs/README.md

erwango, 2018-03-07:
I can help on ST one.
One question: should all the README be all compliant, even when they are high level and lowerlevel ones are compliant? (eg:  ext/hal/st/lib/README)

erwango, 2018-03-07:
It's here for now: 
https://github.com/erwango/zephyr/tree/ext_stm32cube_README_fix
Let me know where I should do the pull request

linkmeyer, 2018-03-08:
While the correct formatting of the README files is important and certainly a real bug, I recommend a separate bug be filed for it as the original intent of this issue (as I read it) is to get the License page updated for the 1.11 release.

Shilpashree3lc, 2018-03-09:
@linkmeyer  I have run the scripts to check for licensing and manually verified them, all the files are in compliant.
The license page is up-to-date.  There are no changes needed to the license page.

carlescufi, 2018-03-09:
@Shilpashree3lc I don't think the license page is up to date? There are many components in the `ext/` folder that are not present in `doc/LICENSING.rst`

Shilpashree3lc, 2018-03-09:
I have completed the work. Hence closing it.

carlescufi, 2018-03-09:
@Shilpashree3lc so there are 2 different issues here as Mark mentioned:

1. Update all the `README` files in `ext/` so they comply with the template
2. Update the `doc/LICENSING.rst` file so that it lists all the components in `ext/`

You obviously cannot be expected to do 1. by yourself, but could we at least update `doc/LICENSING.rst` so that it lists all components in `ext/`, even if it's only the name of the component and the software license?

nashif, 2018-03-09:
> You obviously cannot be expected to do 1. by yourself, but could we at least update doc/LICENSING.rst so that it lists all components in ext/, even if it's only the name of the component and the software license?

This is not needed, the license page is ONLY for components that do not have a license, not for listing everything in ext/. It really has nothing to do with ext/

nashif, 2018-03-09:
the document says that:

> The Zephyr kernel tree imports or reuses packages, scripts and other files that are not covered by the Apache 2.0 License. In some places there is no LICENSE file or way to put a LICENSE file there, so we describe the licensing in this document.

carlescufi, 2018-03-14:
CC @kestewart

nashif, 2018-05-11:
not a bug

MaureenHelm, 2018-05-24:
Added item to release checklist in #6533 
- Update doc/LICENSING.rst for new components that do not have a license

closed 2018-05-24 by MaureenHelm