zephyrproject-rtos/zephyr #6044: native_posix missing features

aescolar-ot, 2018-02-08:
This is a non-exhaustive list of known missing* features in the native_posix board.
\* missing = not (yet) implemented / supported, not (properly) tested.
The purpose being to make it clear for users.

* Kernel:
  * No TICKLESS_KERNEL support
  * TICKLESS_IDLE (only partially tested)
* Devices/drivers:
  * Entropy device (using host random() + seed from command line)
  * Network - ethernet
  * UART
* Subsystems:
  * Console ("pull style", getchar(), getline(), putchar())
  * CPP (no C++ support yet)
  * Logging (not tested at all)
* Ancillary:
  * (Proper) Command line argument parsing

andrewboie, 2019-07-16:
Add CONFIG_HW_STACK_PROTECTION and CONFIG_USERSPACE to this list.
Although off the top of my head I can't think of a good way to do it, it really does depend heavily on hardware. This might need to be a documented limitation.

digiexchris, 2025-05-01:
any updates on this? CONFIG_USERSPACE is required by DYNAMIC_OBJECTS required by k_object_alloc(K_OBJ_THREAD). It silently (and counter to the documentation) does nothing otherwise. In my case, creating a thread dynamically. Given the relatively powerful MCUs we have available to us now, it's nice to be able to use more dynamic memory use, and being able to develop and test much of the application on native_sim is incredibly helpful.

Or, adding documentation to the threading docs to show how to accomplish the same in kernel mode zephyr style would probably suffice for a lot of things. I can malloc manually and get it to work, but that works around all of the nice functions you have otherwise.

aescolar, 2025-05-02:
@digiexchris No updates on USERSPACE support for native_sim, still not planned.

Regarding the DYNAMIC_OBJECTS API and specifically about there being an empty `k_object_alloc()` when that API is disabled, I agree that it seems quite counter-intuitive.
Would you be so kind as to create a separate ticket to track that?

AndreyDodonov-EH, 2025-07-31:
>PTY UART pending feature:
interrupt driven IF

Are there any ongoing/completed efforts/PRs for this?

clamattia, 2025-08-19:
Missing feature RTC-API: https://github.com/zephyrproject-rtos/zephyr/issues/94637

seyoungjeong, 2026-03-01:
Is the issue `Logger backend: control verbosity from the executable command line ; merge the char out function with pre_print` still valid and do we want to have it? If yes I'd like to look into it. @aescolar