zephyrproject-rtos/zephyr #6066: LwM2M: support object versioning in register / discover operations

mike-scott, 2018-02-09:
The only item left to meet OMA LwM2M 1.0 Enabler Static Conformance Requirements (mandatory items) is the following:
LwM2M-CR-012-C-M: Support of Object Versioning in the Object and Object Instances parameter list (Section 5.3.1)

Per the LwM2M tech specification if an object only has 1 version then the version doesn't need to be specified.  And if an object is part of the LwM2M enabler specification then it doesn't need to be specified.  However, objects defined outside of the LwM2M enabler (read as IPSO Temp / Light Control objects) *DO* need to support a version parameter when the object has more than 1 version.

Example Registration of imaginary object 44 w/ a version 2.2:
LwM2M Client “Register” operation: ep=nodename,</1/0>,</1/1>,</3/0>,</44>;ver= “2.2”,</44/0>

As far as I know, all of the objects which are present in Zephyr use the initial (1.0) version.

dleach02, 2019-12-10:
Moving to 2.2

jukkar, 2021-01-20:
@rlubos is this still valid?

rlubos, 2021-01-22:
Not implemented yet.

PR #33466 (2021-03-29): net: lwm2m: Add support for objects versioning and IPSO objects in version 1.1
This PR implements support for object versioning of the LwM2M objects. Initial commit adds object version reporting to the recently introduced Link Format content writer.

Consecutive commits update the IPSO objects implementation to support version 1.1 of their specification (only the objects that actually have an updated specification), which extends the objects with Timestamp resource. Finally, remove the non-standard `LWM2M_IPSO_TIMESTAMP_EXTENSIONS` extension, since it's no longer needed with updated objects.

 [... cut]
closed 2021-03-29 by nashif