zephyrproject-rtos/zephyr #6082: Support the MIPS architecture

nashif, 2018-02-09:
Support MIPS architecture and select boards. In addition and for the purpose of testing support Qemu and verify it passes all kernel tests.

i-and, 2019-03-19:
Hi @nashif,
When do you plan to support the MIPS architecture? Will you support MIPS64?

hongshui3000, 2019-06-12:
When do you plan to support the MIPS R6 architecture?

frantony, 2021-10-30:
@remy-luisant 

I have just updated https://github.com/zephyrproject-rtos/zephyr/pull/30405.

Alas, at the moment very many qemu tests failed (e.g. ``tests/benchmarks/sys_kernel`` just hangs on start), so I have dropped MT76x8 support in the latest update. Tests for qemu malta should be fixed first.

remy-luisant, 2021-10-30:
Can I help somehow?

I do need MIPS support within a couple of months and I'm happy to help it happen.

frantony, 2021-11-08:
@remy-luisant 

I have just updated https://github.com/zephyrproject-rtos/zephyr/pull/30405.

> Can I help somehow?

Could you please take a look at ``tests/subsys/logging/log_api`` and ``tests/subsys/logging/log_msg2`` mips failed tests.

remy-luisant, 2021-11-08:
Thank you, @frantony.

Both me and my partner are quite sick with a bad cold right now, so I have put that on my schedule for this Thursday, SGT. Does this work, or do you need that done sooner?

frantony, 2021-11-08:
@remy-luisant 

> I have put that on my schedule for this Thursday, SGT. Does this work, or do you need that done sooner?

There is no need for haste. Hope you get well soon!

frantony, 2021-11-10:
@remy-luisant 

I have [just pushed new version of MIPS support commits](https://github.com/zephyrproject-rtos/zephyr/pull/30405#issuecomment-965541767). The `tests/subsys/logging/log_api` and `tests/subsys/logging/log_msg2` are fixed now.

At the moment there is only one test problem: `tests/subsys/cpp/libcxx` build failure.

remy-luisant, 2021-11-11:
Brilliant. Thank you.

I am mostly recovered and I am getting the tools set up. I'll give a stab at the testcase. libcxx? That's a curious thing. I wonder why that fails.

The components are here and my PCBs are due to arrive any day now. I should be able to give running this on the X1000E SoC a spin quite soon. I also have a few PIC32 MIPS ICs handy, so that's another thing that I can try fairly easily. Depends on how much coverage we want, I suppose.

frantony, 2021-11-11:
@remy-luisant 

> libcxx? That's a curious thing. I wonder why that fails.

At the moment libc and libcxx from [Zephyr SDK 0.13.1](https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.13.1) are built with feeble CFLAGS/CXXFLAGS which lead to ``relocation truncated to fit: R_MIPS_GPREL16``  link problems. I have rebuilt sdk-ng with more applicable options for libc (see patch in [this comment](https://github.com/zephyrproject-rtos/zephyr/pull/30405#issuecomment-955663713)) to pass the Zephyr tests, but I have no success with libxx rebuild.

> The components are here and my PCBs are due to arrive any day now. I should be able to give running this on the X1000E SoC a spin quite soon.

I have took a look on [Halley2 Devkit](http://www.ingenic.com.cn/?product/id/9/lm/3.html). The Halley2 Coreboard has JTAG pins, but I can't find JTAG pins on carrier board. Does you PCB has JTAG pins connected?

> I also have a few PIC32 MIPS ICs handy, so that's another thing that I can try fairly easily. Depends on how much coverage we want, I suppose.

I have a [PIC32MX Clicker board](https://www.mikroe.com/clicker-pic32mx). But I have never tried to run Zephyr on it.

We already have [Mediatek MT76x8](https://deviwiki.com/wiki/MediaTek#MIPS
) support for Zephyr, e.g. see https://github.com/frantony/zephyr/tree/20210617.mt76x8

MT76x8-based boards are available and very cheap; there is OpenWrt support for these boards; also barebox bootloader (and U-Boot) is available (see https://github.com/frantony/barebox/commits/20200727.mediatek) so we have code for low level SoC initialization.
I use a [Linkit Smart 7688 board](https://labs.mediatek.com/en/platform/linkit-smart-7688) with JTAG connected for Zephyr MIPS experiments.

remy-luisant, 2021-11-11:
> At the moment libc and libcxx from [Zephyr SDK 0.13.1](https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.13.1) are built with feeble CFLAGS/CXXFLAGS which lead to `relocation truncated to [...]
 
It seems like something is trying to use GP-relative addressing and breaks. Please forgive my lack of knowledge of the situation, but why are you disabling the GP optimizations in that patch?

This could be reenabled with a linker script handling the .sbss, .sdata and possibly some other sections to enable space-saving optimizations. You'd also need to add a line or two to your crt0.S to set up the GP register.

When I was messing around with a preliminary stab at a port before I was made aware of this effort I followed pretty much the exact same path with regard to the GP register. I disabled the optimizations and carried on. Eventually, it just became easier to fix it rather than to keep dealing with avoiding it. The space savings weren't amazing, but shorter code was produced with the optimization. I discovered your work soon after that and I saw that you are much further along than I was. Thanks to this it never got much testing. Still, if you wish I am happy to try to integrate the change, it isn't that big.

> I have took a look on [Halley2 Devkit](http://www.ingenic.com.cn/?product/id/9/lm/3.html). The Halley2 Coreboard has JTAG pins, but I can't find JTAG pins on carrier board. Does you PCB has JTAG pins connected?

I designed in JTAG support for the PCB, but given that I'm doing hot air assembly on BGA components here, it might be a few days/tries before things work.

Thanks for letting me know about supported boards! I greatly appreciate it. I might get one, but most likely will just keep on iterating on the X1000E.

frantony, 2021-11-11:
> It seems like something is trying to use GP-relative addressing and breaks. Please forgive my lack of knowledge of the situation, but why are you disabling the GP optimizations in that patch?

According to https://gcc.gnu.org/onlinedocs/gcc/MIPS-Options.html#index-mgpopt ``-mno-gpopt`` not just disables GP-related optimization but GP use. I just want to disable PIC and GP use at all to simplify my work.

> This could be reenabled with a linker script handling the .sbss, .sdata and possibly some other sections to enable space-saving optimizations. You'd also need to add a line or two to your crt0.S to set up the GP register.

First let's make it just work. Optimizations and improvements can be done later.
 
> When I was messing around with a preliminary stab at a port before I was made aware of this effort I followed pretty much the exact same path with regard to the GP register. I disabled the [...]

I'm very happy to see any reasonable changes that make 100% tests pass.
 
> I designed in JTAG support for the PCB, but given that I'm doing hot air assembly on BGA components here, it might be a few days/tries before things work.

It looks like there is out of tree Ingenic openocd support: https://github.com/XBurst/OpenOCD-XBurst/commits/v0.10.0

remy-luisant, 2021-11-11:
> First let's make it just work. Optimizations and improvements can be done later.

A sound approach. 

I have your code cloned, going to try to locate what I'm guessing is a missed GP-disabling flag somewhere.

remy-luisant, 2021-11-11:
Adding the following line to the SDK configuration file might be the solution: `CT_TARGET_CFLAGS="-G0 -mno-gpopt"`

I'll run tests shortly.

frantony, 2021-11-12:
@remy-luisant 

> Adding the following line to the SDK configuration file might be the solution: CT_TARGET_CFLAGS="-G0 -mno-gpopt"

I have rebuit sdk-ng with this patch:

```
--- a/configs/mips.config
+++ b/configs/mips.config
@@ -5,6 +5,7 @@ CT_LOCAL_TARBALLS_DIR="${CT_PREFIX:-${HOME}/x-tools}/sources"
 # CT_LOG_PROGRESS_BAR is not set
 CT_ARCH_MIPS=y
 CT_MULTILIB=y
+CT_TARGET_CFLAGS="-G0 -mno-gpopt"
 CT_TARGET_VENDOR="zephyr"
 CT_BINUTILS_SRC_DEVEL=y
 CT_BINUTILS_DEVEL_URL="https://github.com/zephyrproject-rtos/binutils-gdb.git"
```
after adding discarding the .reginfo section to the linker script 100% tests passed. Could you make a CT_TARGET_CFLAGS PR for https://github.com/zephyrproject-rtos/sdk-ng please?

I have [just pushed new version of MIPS support commits](https://github.com/zephyrproject-rtos/zephyr/pull/30405#issuecomment-966930176).

Please send me a word if you want to add your Signed-off/Tested-by etc to a git commit messages.

remy-luisant, 2021-11-12:
@frantony

I will gladly do the PR sometime tomorrow.

I have written the code for the GP optimization, it builds fine. I get some test failures at runtime, qemu timeouts. I'm content to leave it be for now.

A sign-off would be lovely, thank you.

My PCBs are delayed at customs, so it might be a long while until I can do something on real hardware. Until then, is there anything else I can be of service with? Any rough spots needing polish, etc...?

frantony, 2021-11-12:
> @frantony
> 
> I will gladly do the PR sometime tomorrow.
> 
> I have written the code for the GP optimization, it builds fine. I get some test failures at runtime, qemu timeouts. I'm content to leave it be for now.

I got some test failures (qemu timeouts) too. I have found that my test server is slightly overloaded so I used `-j` option for twister. My server has 16 real CPUs. With `-j 12` and still have qemu timeouts. But with `-j 8` I have no failures.

> A sign-off would be lovely, thank you.

Ok, I need sample of your sign-off or I can copy you sign-off from sdk-ng PR commit.
 
> My PCBs are delayed at customs, so it might be a long while until I can do something on real hardware. Until then, is there anything else I can be of service with? Any rough spots needing polish, etc...?

At the moment `CONFIG_TICKLESS_KERNEL` is not supported. IMHO we can just copy TICKLESS_KERNEL code from `drivers/timer/riscv_machine_timer.c` to `drivers/timer/mips_cp0_timer.c`.

remy-luisant, 2021-11-12:
> > I have written the code for the GP optimization, it builds fine. I get some test failures at runtime, qemu timeouts. I'm content to leave it be for now.
> 
> I got some test failures (qemu timeouts) too. I have found that my test server is slightly overloaded so I used `-j` option for twister. My server has 16 real CPUs. With `-j 12` I still have qemu timeouts. But with `-j 8` I have no failures.

That's the funny thing, those are genuine test failures. The failure mode just happens to be a timeout in each case. I obviously have a mistake somewhere related to the GP register, and 16 testcases fail (8 per endianness) silently. I am a bit surprised it is only 16, but the silent errors are not actually a very inviting start to a debugging session. 
 
> At the moment `CONFIG_TICKLESS_KERNEL` is not supported. IMHO we can just copy TICKLESS_KERNEL code from `drivers/timer/riscv_machine_timer.c` to `drivers/timer/mips_cp0_timer.c`.

Added to my list. 

Do we support XIP? I'm kinda thinking we don't, looking at the linker script, but the configuration option can be toggled in the configuration.

frantony, 2021-11-15:
@remy-luisant

> Do we support XIP? I'm kinda thinking we don't, looking at the linker script, but the configuration option can be toggled in the configuration.

At the moment we use qemu-malta board and rely on qemu's loading mechanism which is very straightforward so there is no need in XIP. On real board we have to use SoC's or bootloader's mechanism which can be tricky and complicated.
On MT76x8 I use barebox/u-boot for starting zephyr.

What bootloader you are planing to use on X1000E?

remy-luisant, 2021-11-15:
I made the PR, as requested. Tickless clock work is set for tomorrow. XIP after, unless you have something else for me to take a look at before that.

> What bootloader you are planing to use on X1000E?

I'll be using a stripped-down instance of Zephyr, since I already have it developed for booting a full version of Zephyr for the Allwinner V3s bootup. XIP buys me the ability to use a secondary SRAM during that process, to make the bootloader's job a bit easier. The X1000E initially only gives you ~10 kilobytes for both data and code. I would rather like to have that ten kilobytes to have as much code as possible. There is a 16 kilobyte chunk probably available for data, if I'm reading the docs right.

frantony, 2021-11-15:
@remy-luisant 

> I made the PR, as requested.

Thanks, now I see https://github.com/zephyrproject-rtos/sdk-ng/pull/411.
The commit looks good there is I can't understand why there is *1 failing check*. Let's wait for reviewers.

remy-luisant, 2021-11-17:
@frantony 

The SDK PR got merged.

The timer is now tickless. It was fairly straightforward to copy what the RISCV timer does. I will be running tests after I catch some sleep. Do you have a preference in how I give you the code?

frantony, 2021-11-17:
@remy-luisant 

The SDK PR is approved, thanks.

> The timer is now tickless. It was fairly straightforward to copy what the RISCV timer does. I will be running tests after I catch some sleep. Do you have a preference in how I give you the code?

Could you please push your branch with tickless timer commits into your github zephyr repo?

frantony, 2021-11-18:
@remy-luisant

> The SDK PR got merged.

Zephyr SDK 0.13.2 with your commit is released: https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.13.2

remy-luisant, 2021-11-18:
@frantony 

Great, thanks for letting me know.

The tickless timer passed all tests. I got busy with PCBs, but do expect my code to hit GitHub shortly, I'll let you know. Do you have anything else that you want to offload, or should I go for XIP after?

I have the X1000Es soldered on and working correctly. I can read hardware registers via the USB recovery mode. I am working on getting JTAG up on that, and then we should be able to test on one more piece of real hardware.

frantony, 2021-11-19:
@remy-luisant 

> The tickless timer passed all tests. I got busy with PCBs, but do expect my code to hit GitHub shortly, I'll let you know. Do you have anything else that you want to offload, or should I go for XIP after?

Please push tickless timer commit to github repo.

My plan is:

1. cherry-pick tickless timer commit;
2. rebase over latest main and run all tests again;
3. brush commits to pass all compliance checks;
4. try Zephyr on a real hardware.

remy-luisant, 2021-11-19:
@frantony 

Here you go.
https://github.com/remy-luisant/zephyr/tree/mips-qemu-malta

Please give it a thorough testing. I am used to fossil and SVN, not git. It took me an embarrassingly long time to get things to align right and it is quite possible I introduced an error somewhere along the way. Let me know if something's not quite right. I'd rather get the practice of redoing it than to have you fix things and me not having learned anything.

frantony, 2021-11-21:
> Here you go.
> https://github.com/remy-luisant/zephyr/tree/mips-qemu-malta
>
> Please give it a thorough testing. I am used to fossil and SVN, not git. It took me an embarrassingly long time to get [...]

Everything is right!

Some cosmetic changes for your commit can be found in this fixup commit:
https://github.com/zephyrproject-rtos/zephyr/pull/30405/commits/3025a3765cf8c112ef163588c0dad5d38ecaca83

Changing ``#if defined(CONFIG_TICKLESS_KERNEL)`` to ``if (!TICKLESS) {`` is a common practice (see 5bc458a0fae521522273302bbb2082d73895e2c9, advantage of ``IS_ENABLED`` described in the 645f867444c2c7391c59fa16b9217006e4fd8c33
 commit message).

If you have no objections I'll squash fixup commit 3025a3765cf8c112ef163588c0dad5d38ecaca83 into your ``timer: Add tickless support for the MIPS CP0 timer.`` commit.

P.S. There a common git commit message recomendation: *do not end the subject line with a period*.

remy-luisant, 2021-11-21:
@frantony

Thank you, your advice is greatly appreciated and will be followed. You are welcome to do whatever you wish with my commits. I'm here primarily to get Zephyr on MIPS working officially, but I'm here also to learn and to contribute.

I have code execution of assembly on the X1000E and the UART working, also from assembly, using the zephyr-sdk toolchain. The UART is a standard 16550, thankfully. Hoping for actual Zephyr execution shortly. JTAG is a bust, however. The interface works correctly on the physical level and the IC is identified correctly. The core constantly reports resets and fails to enter debug mode. This is with the XBurst OpenOCD fork.

Starting on XIP.

remy-luisant, 2021-11-23:
@frantony 

Pointing out a possible issue: I am getting __udivdi3 and __umoddi3 from libgcc pulled into a build of the hello_world sample on the qemu board. They are software implementations of divisions and remainders. As far as I can tell, MIPS has the ability to do divisions and remainders, so this 2.5k worth of math code is not needed.

This inclusion might be prohibitive for both size and speed reasons, at least on some workloads. Might be something for the queue, once either of us has the time for it. Trivial changes to the compiler/linker flags don't seem to make a difference.

frantony, 2021-11-26:
@remy-luisant 

> As far as I can tell, MIPS has the ability to do divisions and remainders

Yes, it has. I suppose that libgcc divisions and remainders stuff is used at least for 64-bit numbers.

remy-luisant, 2021-11-26:
@frantony

I just checked, you are correct. All uses are in 64-bit integer processing. Not gonna be easy to make that go away.

frantony, 2021-11-26:
@remy-luisant 

The good news is after reverting 70dd3d6b0deec17eab6e0888d458b7e5aa1481da (see https://github.com/zephyrproject-rtos/zephyr/issues/40569) GitHub CI reports that MIPS port pass all tests.

There is only two problems:

  * *Compliance Checks Failing*; I have to add some lines into CODEOWNERS;
  * *Scan code for licenses Failing*; there are several core files introduced in original Imagination port not apache-2.0 licensed. I need at least several days to arrive at a solution.

remy-luisant, 2021-11-26:
@frantony

All right. Good to know. Thanks for keeping me posted.

I found some time to push XIP a bit, but mainly working on getting things going on the real hardware, where size is an issue even with hypothetical XIP. Likewise, we are looking at quite a few days, given all else I am working on right now.

remy-luisant, 2021-11-27:
@frantony 

XIP is done. https://github.com/remy-luisant/zephyr/commit/5e791eb6f248dbc7e477d8908de99c89e34864f2

Let me know if you need me to make any changes. I expect the dts changes might be controversial, though they match the circumstances of my use case pretty closely. Let me know if there's a better way about this. As always, I'm happy to learn. 

I think the two outstanding items you mentioned are best left to you. Anything else I can lend a hand with?

frantony, 2021-11-28:
@remy-luisant 

> XIP is done. [remy-luisant@5e791eb](https://github.com/remy-luisant/zephyr/commit/5e791eb6f248dbc7e477d8908de99c89e34864f2)
> 
> Let me know if you need me to make any changes. I expect the dts changes might be controversial, though they match the circumstances of my use case pretty closely. Let me know if there's a better way about this. As always, I'm happy to learn.

I have some experience with Ingenic JZ4755 and JZ4775 MIPS SoCs. IFAIR JZ4755 and JZ4775 MIPS use bootrom. After reset program in bootrom starts, checks dedicated chip input pins to determine boot mode (e.g. SD/MMC, NOR/NAND flash and even USB). Next bootrom code loads program from SD/MMC/NOR etc into RAM and starts it.
qemu-malta work in a more simple way. qemu-malta board has emulated parallel NOR flash at 0xbfc00000. If qemu started with ``-bios <raw binary file>`` option then qemu copy the ``<raw binary file>`` into NOR flash and CPU starts at 0xbfc00000.
If qemu started with ``-kernel <ELF file>`` (we use this method in current MIPS support branch) then qemu loads ELF into RAM, copies short trampoline code to 0xbfc00000 which just simply jumps to ELF entry point address and CPU starts at 0xbfc00000. 

So to fully test your XIP code for qemu-malta at least two things should be done:
* add true boot NOR flash record into qemu-malta dts file;
* use ``-bios <raw binary file>`` option for qemu.
 
> I think the two outstanding items you mentioned are best left to you. Anything else I can lend a hand with?

Which boot mode for X1000 you use? If you use boot mode which available in JZ4770/JZ4755 then I could try to test zephyr on JZ4770/JZ4755.

Can you publish your X1000 soc/board code?

remy-luisant, 2021-11-28:
@frantony 

I will gladly make the changes you requested.

The X1000E also has a bootrom, reads strapping pins and boots in different modes from there. I designed the PCB to be able to change them via a jumper. I will be using the USB recovery mode and the SPI mode. I fully intend to publish everything I have, once I have it. I am mainly looking for work and trying to make this into a product is a side-project. Sadly, the pace of my work is slow and very uneven as interviews and other priorities come and go. Apologies. I also have been working on reducing the ROM usage, since I need that for my use case. If there's something you wish me to prioritize I am very happy to do so and I am happy to take deadlines too.

I also have the GP optimization working and passing all tests without XIP. XIP has uncovered a problem, all logger tests are failing. The failure is total, but somewhat encouragingly limited to only one subsystem. I have the cause diagnosed and some logger tests passing with a messy fix. There also is a workaround that makes the problem go away at the cost of reducing the effect of the optimization. A clean and complete fix eludes me. I'll continue working on it, on and off.

frantony, 2021-11-28:
@remy-luisant 

> I also have the GP optimization working and passing all tests without XIP. XIP has uncovered a problem, all logger tests are failing.

Please try to apply your GP optimization commits on top of my recent branch (2964f64) and run logger test with XIP enabled.

remy-luisant, 2021-11-28:
@frantony 

I am afraid to report that it didn't work.

The failures are caused by the issues relating to the linker script. The logger constants are small enough to qualify for inclusion in the small data space, which is in the RAM (and in the ROM as well, before being copied to RAM). The compiler will place them there by default. However, the section assignment is overriden from C to place them in a dedicated logger constant section, which is solely in the ROM. The code which uses the logger constants still wants to access them via GP, simply because of the small size of the constants, nevermind the location. The GP points into the RAM and has a reach of 64 kilobytes. If the RAM and the ROM are more than 64 kilobytes apart, the build will fail.

The messy solution involves intercepting the offending section before the linker script includes common-rom.ld, putting the logger constants into the small data section and, at runtime, copying the logger constants along with small variable data into the RAM to be GP-reachable. There are a couple of boards that give the logger constants special treatment, so this isn't without precedent. However, doing all that trades off RAM for ROM. (Doing the same for the variable data is always a net win.) I have implemented that and it passes a few more testcases, but the tradeoff might be unwelcome. The workaround reduces the size of the qualifying small objects down to 4 bytes, so that the logger constants no longer qualify. This inhibits some optimizations I'd like to see, like the ones seen on Nios2 which has a GP register as well. Ideally, I'd like the constants to stay in the ROM and the size of the qualifying small variable data to be able to be adjusted upward. GCC flags aren't helpful here, the only useful tunable is the size cutoff.

Without XIP everything happens to fall within a small enough span that this doesn't matter and the constants can still be reached. This doesn't mean that the right thing happens, the problem is still there, only masked. Combining GP and XIP exceeds the reach.

As far as XIP alone, I have made the required changes to the dts and to the qemu command line. ARM64 already does a similar thing that we needed. The last hurdle is getting a reasonably-sized raw binary. Currently, the raw binary happens to span 1 gigabyte, which is obviously wrong. I think I have it diagnosed, working on a solution.

remy-luisant, 2021-11-29:
@frantony 

All right, I managed to get a degree of XIP execution with the proper qemu setup and with your latest branch. I am testing on big endian for now, since the other endianness currently fails. I presume that one will be fixable by adjusting the process of the binary creation. I will deal with that one later.

The current problem is that a syscall happens and it vectors to 0x80000180, the standard MIPS exception vector. I do not believe this is intended. I can add code to reroute this into our vectors. This would be a change to the EBase CP0 register, most notably. This doesn't strike me as a XIP-only change. At least some real hardware would want this too. I believe you mentioned that you were testing on real hardware in the past, so I'm guessing that the changes to that might already exist. In the interest of avoiding a duplicate effort, I was wondering if you are taking care of it. If not, I'll get that taken care of.

remy-luisant, 2021-12-01:
@frantony

I got further. I made the interim change to the EBase and ST_BEV. All tests pass with XIP on the be qemu target now, using raw binaries. I'm going to clean up and push to a branch shortly. One last issue: I'm guessing to make le work a byte order reversal is needed for the raw binary to work. If you have code that sets up EBase and BEV, I will replace my setup code with yours.

Using this setup, I compiled some code for the hardware. The raw binary executables that I ran on the SoC crash before the boot banner, but patching the reset code to light an LED shows that code does execute for at least a little while. Progress!

frantony, 2021-12-04:
@remy-luisant 

> One last issue: I'm guessing to make le work a byte order reversal is needed for the raw binary to work.

Please see this recipe: https://github.com/saschahauer/barebox/commit/8ce2aba5c51e27da3dcb80825682b3aba3c3b627#diff-45f753be547ee19b1eb7c12bd03bb1c3fddd844ac8f66b91e0f6a2b4e4018a11R18

remy-luisant, 2021-12-04:
@frantony 

Thank you for the recipe. I actually have the swap part figured out, the problem is is not being sure how to integrate it into the cmake build. Changing the flags for objcopy to add --reverse-bytes applied to an intermediate file, which caused issues down the road. I would wish for it to apply solely to the final binary. For running on hardware I rely on the build-in swapping of the upload program, but I would love to test le on qemu too.

I have execution on real hardware all the way until the boot banner, which implies surviving at least a syscall exception and a thread switch. The UART busy-waits forever, probably a bug in my dts configuration. I have a workaround in the works, which should at least make printks work. Debugging with one RGB LED gets old fast. I am pretty close to reaching main() now. I did require an SoC custom linker script and I'm still at 95% of "rom" taken, without even the proper payload included yet. I will be looking at the GP stuff again after the code cleanup, since it actually might be the difference between being able to make it fit and not.

It's looking good. I will be publishing my changes probably next week, once I clean them up a bunch. I'm not sure how much of that you would want to integrate, but at least the option will be there.

frantony, 2021-12-05:
@remy-luisant 

> Thank you for the recipe. I actually have the swap part figured out, the problem is is not being sure how to integrate it into the cmake build. Changing the flags for objcopy to add --reverse-bytes applied to an intermediate file, which caused issues down the road. I would wish for it to apply solely to the final binary. For running on hardware I rely on the built-in swapping of the upload program, but I would love to test le on qemu too.

Qemu-mipsel binary image octet swapping is just a qemu caprice. I can't remember any octet swapping necessity on real bi-endian hardware.
 
> I have execution on real hardware all the way until the boot banner, which implies surviving at least a syscall exception and a thread switch. The UART busy-waits forever, probably a bug in my dts configuration.

Please see an example of Ingenic dts UART node from barebox https://github.com/saschahauer/barebox/blob/master/arch/mips/dts/jz4780.dtsi#L17
Be careful! Barebox (and linux!) uses physical addresses in ``reg``. On MIPS physical addresses always differ from virtual (CPU) addresses so translation is needed. In zephyr (just for simplicity) there is no phys-to-virt dts reg addresses translation.

> I have a workaround in the works, which should at least make printks work. Debugging with one RGB LED gets old fast. I am pretty close to reaching main() now. I did require an SoC custom linker [...]

Please try to use latest https://github.com/frantony/zephyr/tree/mips-qemu-malta as a base for your changes.

remy-luisant, 2021-12-05:
@frantony 

I did have byte swapping issues with the actual hardware initially as well, when trying to read registers via the USB recovery software. I don't remember the exact circumstances, but I can document it and the fix I made, if needed and if my recollection of having the problem is actually correct. I might easily be wrong on this one, since you say that it shouldn't be required.

As far as the UART: The only problem is the configuration step of the driver initialization. Commenting out the configuration and doing it myself in assembly works just fine. The addresses taken from the dts are correct and the driver is otherwise working. I have both the baud rate and the clock speed specified correctly. The oscilloscope shows no activity after the configuration is allowed to run. This is in line with the driver getting stuck in poll_out. Using the workaround and with printks now working my debugging went much faster.

I arrived at main(), finally. I had to enable time slicing, something I had disabled for space reasons previously. My, apparently wrong, assumption was that the main thread would continue to execute, which doesn't seem to happen. I don't have any other threads in the system, aside from possibly the idle thread, which shouldn't be scheduled in due to having a lower priority. I will look into it. I have a few more kilobytes to spare now, thanks to a new slice of memory that I found, so I enabled the time slicing and things work. A quick survey of my changes shows me that my work is mostly board-specific, so it actually won't be that hard to clean it up. Thank you for letting me know there is an update, it is greatly appreciated, as always.

frantony, 2021-12-09:
@remy-luisant

I have updated MIPS support PR and requested review (see 
https://github.com/zephyrproject-rtos/zephyr/pull/30405#issuecomment-989766099). 

Please use my latest MIPS support branch! There is a small but important change: I have disabled hard-float (enabled in Zephyr MIPS SDK by default) because current MIPS save/restore context routines don't take care on FPU (CP1) registers.

remy-luisant, 2021-12-09:
@frantony 

Thank you, will do. I was stuck on job hunting and power supply design work for the last few days, so I had no chance to do anything since my last comment here. Apologies. I think I have the byte swapping added to CMake in a way that won't cause issues. but I didn't even have the chance to run it once yet.

frantony, 2021-12-10:
@remy-luisant

Please try this patch.

```
diff --git a/arch/mips/core/isr.S b/arch/mips/core/isr.S
index 19764588c7..dd96f786ab 100644
--- a/arch/mips/core/isr.S
+++ b/arch/mips/core/isr.S
@@ -270,6 +270,8 @@ no_reschedule:
        OP_LOADREG k0, ESF_O(epc)(sp)
        mtc0 k0, CP0_EPC
        OP_LOADREG k1, ESF_O(status)(sp)
+       li k0, ST0_EXL
+       or k1, k1, k0
        mtc0 k1, CP0_STATUS
        ehb
 
diff --git a/arch/mips/include/mips/mipsregs.h b/arch/mips/include/mips/mipsregs.h
index 7b555e7c65..293ccf5acb 100644
--- a/arch/mips/include/mips/mipsregs.h
+++ b/arch/mips/include/mips/mipsregs.h
@@ -17,6 +17,7 @@
 
 /* CP0_STATUS bits */
 #define ST0_IE         0x00000001
+#define ST0_EXL                0x00000002
 #define ST0_ERL                0x00000004
 #define ST0_IP0                0x00000100
 #define ST0_BEV                0x00400000
```
qemu works fine without it but real hardware may need it for correct work.

remy-luisant, 2021-12-14:
@frantony 

Sorry for the absence. All my electronics/embedded stuff is sadly just a hobby/portfolio thing and I needed to focus elsewhere for a while. I finally have a potential employer lined up now, so I can take a look at this again a little bit more.

What hardware are you running this on and what are the symptoms of the problem? Possibly related: As part of my XIP exploration I developed a patch that is required to boot on machines where SRAM doesn't start at 0x80000000, a change to the linker script and to one of the CP0 registers, if I recall. You might be running into that if you are on an R2 or newer machine. I would get crashes on the X1000E, since neither of the two SRAMs is at that location and there was an unconditional copy to 0x80000000.

frantony, 2021-12-14:
> @frantony
> 
> What hardware are you running this on and what are the symptoms of the problem? 

I use this: https://www.seeedstudio.com/LinkIt-Smart-7688.html

The symptom is very simple: Zephyr just hangs without any message.

Now I use a more appropriate patch to make Zephyr work on LinkIt-Smart-7688:

```
diff --git a/arch/mips/include/mips/mipsregs.h b/arch/mips/include/mips/mipsregs.h
index 7b555e7c65..293ccf5acb 100644
--- a/arch/mips/include/mips/mipsregs.h
+++ b/arch/mips/include/mips/mipsregs.h
@@ -17,6 +17,7 @@
 
 /* CP0_STATUS bits */
 #define ST0_IE         0x00000001
+#define ST0_EXL                0x00000002
 #define ST0_ERL                0x00000004
 #define ST0_IP0                0x00000100
 #define ST0_BEV                0x00400000
--- a/include/arch/mips/arch.h
+++ b/include/arch/mips/arch.h
@@ -25,7 +25,7 @@
 #define OP_LOADREG lw
 #define OP_STOREREG sw
 
-#define CP0_STATUS_DEF_RESTORE (ST0_IE)
+#define CP0_STATUS_DEF_RESTORE (ST0_EXL | ST0_IE)
 
 #ifndef _ASMLANGUAGE
 #include <sys/util.h>
```

remy-luisant, 2021-12-16:
@frantony 

Quick question. The linker script currently in use is partially relying on config variables, which are being set up elsewhere. A few other arches use dts for their SRAM/FLASH memory addresses.

Do you have a preference?

frantony, 2021-12-16:
@remy-luisant 

> Quick question. The linker script currently in use is partially relying on config variables, which are being set up elsewhere. A few other arches use dts for their SRAM/FLASH memory addresses.
> 
> Do you have a preference?

I suppose that using dts for  RAM/ROM memory addresses is a good idea.

I have updated MIPS branch several hours ago, please use it.

remy-luisant, 2021-12-17:
@frantony 

All right, done. All except two tests passed the run with XIP forced on. This is due to libcxx failures again. It actually builds and fails at runtime, in test_exception. Otherwise, both be and le work fine. Once I have those two I will base myself on your latest work and put it up.

frantony, 2022-01-11:
@remy-luisant 

Could you please publish your XIP branch?

remy-luisant, 2022-01-12:
@frantony 

Apologies. I am working on it, on and off, but it isn't ready yet.

All tests now pass in XIP mode, but the fix to the C++ exceptions is unacceptable since I changed the global Zephyr build files to make it work. I think I have a proper fix, but it needs testing. 

The problem was that the conversion from ELF to BIN discarded the section that MIPS needs for exceptions and filled the gaps with 0xff. The discard happens on the command line during the file format conversion process, not in the linker script. Removing the discard from the linker command line fixes this, but I obviously cannot publish something like that since other arches presumably expect it to happen. My current idea is to use the linker script to put the C++ exceptions section into the .rodata section of the ELF file, to hopefully hide it from the discarding that happens during the conversion. That is the plan for the next couple of days, at any rate.

closed 2022-01-19 by nashif