zephyrproject-rtos/zephyr #6083: Bluetooth: Regression in MESH tests

MariuszSkamra, 2018-02-09:
Seems that patch ad7c9d3d76aef3b6298fbca7c27a3e202f777d70 introduces regression in MESH tests.

Affected TCs:
MESH/NODE/IVU/BI-01-C
MESH/NODE/IVU/BI-02-C
MESH/NODE/IVU/BI-04-C
MESH/NODE/IVU/BI-05-C
MESH/NODE/IVU/BI-06-C
MESH/NODE/IVU/BV-01-C
MESH/NODE/IVU/BV-03-C
MESH/NODE/IVU/BV-04-C
MESH/NODE/IVU/BV-05-C
MESH/NODE/KR/BI-05-C
MESH/NODE/KR/BV-02-C
MESH/NODE/KR/BV-03-C

Test Cases are failing, because PTS is not receiving Secure Network Beacons.

jhedberg, 2018-02-09:
@MariuszSkamra do you have any more details of what's going on? Does this look like a host or controller issue? (e.g. do you see the host sending the right advertising HCI commands, etc?)

jhedberg, 2018-02-09:
One thing to note what makes secure network beacons "special", unlike mesh network packets which are configured by default to 3 transmissions (3 advertising events), beacons only have a single transmission. This means that the host-side advertising thread will stop the advertising much sooner. The formula used for the duration is as follows:

```
duration = (BT_MESH_ADV(buf)->count + 1) * (adv_int + 10);
```

In the case of secure network beacons buf->count is 0 and adv_int is 20, so the above should yield a duration of 30ms.

jhedberg, 2018-02-09:
@cvinayak based on the description from @MariuszSkamra it sounds like with the new controller settings the controller doesn't succeed in sending out a single advertising event if the advertising is stopped in 30ms.

cvinayak, 2018-02-10:
If there is scanning active, then my assumptions are that with an optimized scan window changes due to https://github.com/zephyrproject-rtos/zephyr/commit/ad7c9d3d76aef3b6298fbca7c27a3e202f777d70, starting an advertisement event is delayed upto twice the scan window in comparison to older implementation wherein the first advertising event happened to be delay only by one scan window( the current active one).

jhedberg, 2018-02-10:
This is mesh, so continuous scanning with 10ms window is always active. The only exception is if we're acting as a Low Power Node, but that's not the case here. Twice the scan window would mean max 20ms delay, so the 30ms wait that the host does before stopping advertising should be enough, shouldn't it? @cvinayak as long as we're stuck with standard HCI, would you recommend some improvement/fix to the way that we calculate the advertising duration on the host side (time between receiving command complete for write_adv_enable and disabling advertising). As I mentioned earlier, given COUNT desired retransmissions and ADV_INT advertising interval the host currently sets the duration as (COUNT + 1) * (ADV_INT + 1)

jhedberg, 2018-02-10:
Btw, I suppose this all shows that the Mesh vendor extension can't come soon enough.

commit af13d6d5 (2018-02-12): Bluetooth: controller: Fix soft latencies in periodic ticker

Fix, add the missing code for the removal of any accumulated
soft latencies or negative drift ticks when scheduling next
interval expiry with added laziness.

Typically a first interval would accumulate soft latencies
and this has to be removed if the interval is rescheduled
with any added laziness (scheduled to the next soft real
time interval).

Example, scan windows block any new scheduling until the end
of the window, adding latencies to any soft real time ticker
expiry which should try to execute as early as possible after
the  [... cut]
commit 1913a9f5 (2018-04-30): Bluetooth: controller: Fix soft latencies in periodic ticker

Fix, add the missing code for the removal of any accumulated
soft latencies or negative drift ticks when scheduling next
interval expiry with added laziness.

Typically a first interval would accumulate soft latencies
and this has to be removed if the interval is rescheduled
with any added laziness (scheduled to the next soft real
time interval).

Example, scan windows block any new scheduling until the end
of the window, adding latencies to any soft real time ticker
expiry which should try to execute as early as possible after
the  [... cut]
closed 2018-02-12 by jhedberg