zephyrproject-rtos/zephyr #6086: Ideas/requirements for improved TLS support in Zephyr

pfalcon, 2018-02-09:
There is of course TLS support in Zephyr (using mbedTLS), exposed e.g. via net_app client/server API. However:

* I keep hearing from security people that there's something (more) to be done.
* There is a TLS API for sockets being designed: https://github.com/zephyrproject-rtos/zephyr/issues/5900

So, that gives a chance to see if there is something can be improved (or maybe even redone).

pfalcon, 2018-02-09:
One matter which concerns me is getting the right model of TLS-related resources structuring/allocation/sharing. Let me try to dump my mind on that, as an "abstract" model, and more knowledgeable parties can add to/criticize that.

1. So, at the basic level, we have system-wide resources. A good example is entropy source. That's definitely a "global" resource, as examplified by Linux /dev/random.
2. Above that, we'd have "TLS configuration contexts". The "context" word may be somewhat ambiguous here, as specific TLS state for a particular communication channel is also called "context". We're talking about "TLS configuration contexts", which is a means to reuse TLS configuration settings across *different* communication channels. The idea here is that, as security people keep telling us, configuration TLS settings "from scratch" is error prone, and thus insecure. So, a good idea is to have a preconfigured set of such settings, so users can refer to it "symbolically".
2.1. In general case, there should be support for several configuration contexts, for example: one context for OTA updates, another context for DTLS services (e.g. LWM2M), another context for TLS services (can be split in 2 either - for builtin server, vs accessing 3rd-party sites on the Internet).
2.2. At the same time, for simple apps (e.g. samples), just a single, default context would be enough.
2.3. The default context is also a cornerstone of "secure by default" strategy: Zephyr comes preconfigured with settings which are free from known weaknesses.
2.4. Other configuration contexts, apps would need to create at runtime. Likewise, some settings, including in the default context, may require runtime update (e.g. setting root certificates).
2.5. So, in general, configuration contexts should be runtime-configurable, but policy for that likely also should be a configurable - from a completely "locked-down" system, to a fully enabled, dynamic configuration.
3. Based on the above, to create a TLS stream (https://github.com/zephyrproject-rtos/zephyr/pull/5985), we'd need following parameters: a) underlying transport stream, b) context (can be NULL for default). Actually, I'd say there should be 3rd param: c) server_or_client flag. (I may be biased here, because I got used to Python API "for humans", which is structured this way.) Unfortunately, for mbedTLS, server_or_client flag is a part of general TLS configuration. Well, fairly speaking, client vs server usecases are sufficiently different, so that could make sense. But that also means that we'd need to have 2 default contexts - one for client, and one for server. Actually, that covers only the TLS case, we'd need another pair for DTLS. So, 4 default contexts, no accessing by "NULL", instead by explicit identifiers.

Beyond that, we'll need:

4. TLS test mode. I heard valuations up to "we should remove (or was it just disable?) test RNG (==entropy source)". Well, unlikely, because development/testing is very important scenario. But structuring it in such a way that it's unmissably clear that it's the test mode - may be helpful. So, that would be a global flag, which would enable (maybe even force) test RNG, enable RC4, etc. support for all default contexts, also print "INSECURE TLS!!11" warning on boot, also on each creation of TLS stream. The caveat would be console output, so maybe that would be forced on too. There're definitely caveats and edges to consider.

pfalcon, 2018-02-09:
@d3zd3z : FYI. Comments are welcome. (I'm going to add more info later too.)

d3zd3z, 2018-02-09:
Aside from session configuration, the other important aspect of TLS, which is authentication. Web-type TLS generally uses server certs with the client having a list of trusted CA (certificate authorities). This list itself, it typically a few hundred KB, and not practical for most IoT devices. In addition, we will want the client to authenticate with the server, which will require one of three approaches:

1. Certificates. This can either be a limited set of root CAs, or a separate CA list, where the vendor manages the CAs for a configuration. For testing, it is probably best to have the root cert and client certs just in the build.
2. Bare pub/priv keys. This is specified as part of LWM2M.
3. Pre-shared keys (PSK). There are shared keys that are deployed and delivered to the targets. Despite the disadvantages, this is probably the most common scenario with these protocols. For example, the implementations of LWM2M I've looked at only implement this. It is also specified by Thread (although in a very different way).

Each of these has its own TLS ciphersuites to support it, and involves configuration, and use. Which is appropriate depends on the particulars of the device (they involve different tradeoffs).

lpereira, 2018-02-09:
For inspiration, something like libtls from [OpenBSD would be a good start](https://www.openbsd.org/papers/linuxconfau2017-libtls/). [Man page](https://man.openbsd.org/tls_init) (related pages are linked).

pfalcon, 2018-02-13:
>  the right model of TLS-related resources structuring/allocation/sharing
> 1. ...
> 2. ...
> ...

Now let's see what we have now:

1. net_app: https://github.com/zephyrproject-rtos/zephyr/blob/master/include/net/net_app.h#L365 . So, there's a struct which contains a flat list of all things mbedtls. That structure is part of `struct net_app_ctx`. That is described as "Network application context.", but there's a nice ambiguity there. Usually when "application" is being talked, that's a singleton object containing global state for an entire program (e.g. Qt's qApp, etc.). Well, that's not it. `struct net_app_ctx` is actually a client for a particular client or server endpoint. So, each such contains complete disjoint set of all possible mbedTLS config, nothing is shared.
2. https://github.com/zephyrproject-rtos/zephyr/blob/master/include/net/http.h#L529 . So, we have bunch of config encapsulated into `struct net_app_ctx`, but now we want to wrap and leak that thru each API which uses `struct net_app_ctx` underlyingly? That doesn't seem like a scalable approach.
3. In https://github.com/zephyrproject-rtos/zephyr/pull/5985/files#diff-1f973aa6f9b94237c2a14c84da49562cR36 , a similar flat design is prototyped.

So, we don't have an existing design which goes along the lines of https://github.com/zephyrproject-rtos/zephyr/issues/6086#issuecomment-364553313 , but apparently we should.

pfalcon, 2018-02-13:
> For inspiration, something like libtls from OpenBSD would be a good start. Man page (related pages are linked).

Yes, this is the same as mentioned https://github.com/zephyrproject-rtos/zephyr/issues/5900#issuecomment-362289985

Here're my points of how we differ form them:

From https://man.openbsd.org/tls_init#DESCRIPTION :

> The tls family of functions establishes a secure communications channel using the TLS socket protocol.

So, their scope is *only* TLS communications, so they're free to convenience-alize, hide, or mythologize (choose term per your liking) how these communications are done.

On our side, we try to find a common way to communicate over TCP and TLS. Objectively quanifiable way to do that is to do as little as possible to achieve that. On top of that, we can add (potentially unlimited, as based on subjective criteria) convenience APIs - *later*.

Specifially, addressing @GAnthony's concerns in https://github.com/zephyrproject-rtos/zephyr/pull/5985#issuecomment-363546955 :

> It would look more consistent if all the operations where subsumed under a common API.
> The link provided in #5900 (https://www.openbsd.org/papers/linuxconfau2017-libtls/#slide-16) seems to be a reasonable API that tries to address the problem at hand. Maybe it can be generalize somewhat, but something like this seems more intuitive.

https://www.openbsd.org/papers/linuxconfau2017-libtls/#slide-16 lists API consisting of:

~~~
tls_init()

tls_config_new()
tls_config_set*()

tls_client() or tls_server()
tls_configure()
tls_connect*() or tls_accept*()
tls_handshake() ]
tls_read()/tls_write()
tls_close()
~~~

So, let me explain why tls_connect() or tls_accept() aren't part of the API I propose. It's simple: they're not part of TLS operations, prooflink: https://www.ietf.org/rfc/rfc5246.txt . That RFC doesn't specify a way how the underlying connection to a server is established, because that establishment is handled by e.g. TCP means, which is described in the separate RFC (and TLS is not limited to TCP, any byte-oriented reliable peer-to-peer communication channel would do).

So, if there're standard BSD Socket APIs connect() and accept() to handle that part for both TCP and TLS, then it *at least* superfluous to have tls_connect() and tls_accept(). There may be reasons to add them, but as mentioned above, they are subjective and go both ways (pro and cons). We design the API for resource-constrained devices, and the minimality is more or less objective criteria (you can simplify things only so much while still supporting needed featureset), and both go in the same direction, that's why I propose to stick with that design criteria (for now).

pfalcon, 2018-02-13:
Now commonalities:

> tls_client() or tls_server()

Well, that's pretty much what I saw talking about, except I had a view of that being a param to a func, e.g. `tls_new(..., bool server_side)`, while they split those into 2 functions.

Except that per https://man.openbsd.org/tls_client.3, tls_client() and tls_server() don't take any args. That's because they're concerned with only TLS, so can pretend it stands in the vacuum, while we explicit treat TLS just as a layer on another stream (like TCP), so accept that as an arg.

They also have a separate step to configure a connection, using

~~~
tls_configure(struct tls *ctx, struct tls_config *config); 
~~~

But for us, it's natural to accept configuration as a constructor arg (well, that's how mbedTLS works). And as discussed above, unfortunately, there's no separate `bool server_side` setting, it's a part of the general config.

That leads us to following constructor:

~~~
int zstream_tls_init(struct zstream_tls *p, zstream sock, struct tls_config *tls_conf);
~~~

Which is equivalent to libtls' API, modulo param and calls regrouping, and different initial requirements of hiding vs not hiding the underlying connection.

pfalcon, 2018-02-13:
Based on the above, I'd like to propose following structuring (based on https://github.com/zephyrproject-rtos/zephyr/pull/5985/files#diff-1f973aa6f9b94237c2a14c84da49562cR36):

~~~
+	mbedtls_entropy_context entropy;
+	mbedtls_ctr_drbg_context ctr_drbg;
~~~

This would be system-global settings: entropy is system-wide, ctr_drbg "stretches" entropy using crypto-secure RNG, given that underlying entropy is system-wide, stretched entropy is apparently system-side too. Caveat: using them in such way requires synchronizing access to them for multiple threads.

~~~
+	mbedtls_ssl_config conf;
+	mbedtls_x509_crt cacert;
+	mbedtls_x509_crt cert;
+	mbedtls_pk_context pkey;
~~~

This will be in `struct tls_config`, potentially shared among different streams. (Internally, 
mbedtls_ssl_config would keep references to other structures - to be confirmed).

~~~
+	mbedtls_ssl_context ssl;
~~~

This is the only thing which needs to go into zstream subclass for TLS - mbedtls_ssl_config is referenced from it (confirmed).

Comments are welcome, otherwise, I'll try to prototype it in the scope of https://github.com/zephyrproject-rtos/zephyr/pull/5985 .

d3zd3z, 2018-02-14:
A few points:

1. x509 is probably only useful in larger systems, and we will both need a trusted cert store, as well as client side certificates to make this meaningful.
2. We should support PSK, since that seems to be commonly used in IoT.
3. I think we should also think about DTLS/UDP as well. It is supported in mbed TLS, and will indeed be used by COAPS/LWM2M.
4. I wouldn't consider session resumption to be just an optimization. Without it, connections are likely to take too long, and people will end up disabling TLS entirely, which we don't want.

pfalcon, 2018-02-16:
@d3zd3z : Thanks for comments!

> 1. x509 is probably only useful in larger systems, and we will both need a trusted cert store, as well as client side certificates to make this meaningful.

Yes, but that's how TLS is used in the real world (millions on publicly accessible production hosts), so: a) we can't ignore it; b) we can't implement it completely in Zephyr so far at all (no real-time clock support), and most systems won't be to support it fully anyway. So, work on that would be a typical long winding road of compromises and successive approximations.

> 2. We should support PSK, since that seems to be commonly used in IoT.

Sure. Generally, I think we should "support" everything that mbedTLS supports, and if we add or switch to an alternative, we should support its everything too. That's why this ticket starts with trying to find a generic structuring of TLS configuration, and decoupling TLS connection establishment from TLS configuration (in the same vein as #5900 decoupled TLS connection usage from its establishment). If we find generic configuration interface, we'll be able to support any particular configuration too (perhaps, with slight adjustments of the structure).

> 3. I think we should also think about DTLS/UDP as well. It is supported in mbed TLS, and will indeed be used by COAPS/LWM2M.

Yes. The current situation, as I see it, is following: I don't think that ideas discussed here would be applied to existing codebase (net_app, which support DTLS). Instead they intended to be applied to new socket-based API, #5900. And that starts with just wrapping TCP sockets to keep the scope of work manageable. The idea is that the design will be possible to apply to UDP/DTLS too, perhaps with minor adjustments. So, if there're concerns that config structure as proposed in https://github.com/zephyrproject-rtos/zephyr/issues/6086#issuecomment-365347089 and above wouldn't fit DTLS, please share these points. I don't see any immediate problems like that (but specifically postpone detailed consideration of that until later to avoid analysis paralysis situation, given that the working prototype needs to be presented at the beginning of March).

> 4. I wouldn't consider session resumption to be just an optimization. Without it, connections are likely to take too long, and people will end up disabling TLS entirely, which we don't want.

"Without it, ... take too long, with it, takes less" is a definition of "optimization" for me. That said, I'd think that the same comment as for p.2 applies: as long as we have a general and generic TLS configuration setup, we should be able to support features like stored sessions. If you think that their support would affect the general configuration API as proposed above, please elaborate what may need changes. Thanks!

pfalcon, 2018-02-27:
Now (actually second iteration of) the implementation of these ideas is available in https://github.com/zephyrproject-rtos/zephyr/pull/5985, as a commit "tls_conf: Convenience API for TLS configuration settings" (current direct link https://github.com/zephyrproject-rtos/zephyr/pull/5985/commits/b14dbe1d67a532730dd7bbf7bb7592ecc46cd563 , bit will be rebased).

jukkar, 2019-05-22:
This seems to be related to net_app API which was removed recently -> closing.

closed 2019-05-22 by jukkar