zephyrproject-rtos/zephyr #6093: [Coverity CID: 182778] Error handling issues in /samples/net/sockets/dumb_http_server/src/socket_dumb_http.c

Shilpashree3lc, 2018-02-10:
Static code scan issues seen in File: /samples/net/sockets/dumb_http_server/src/socket_dumb_http.c
 Category: Error handling issues
 Function: main
 Component: Samples
 CID: 182778
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182778:  Error handling issues  (CHECKED_RETURN)
/samples/net/sockets/dumb_http_server/src/socket_dumb_http.c: 73 in main()
67     		/* Discard HTTP request (or otherwise client will get
68     		 * connection reset error).
69     		 */
70     		while (1) {
71     			char c;
72     
>>>     CID 182778:  Error handling issues  (CHECKED_RETURN)
>>>     Calling "recv(client, &c, 1U, 0)" without checking return value. 
>>> This library function may fail and return an error code. [Note: The 
>>> source code implementation of the function has been overridden by a 
>>> builtin model.]
73     			recv(client, &c, 1, 0);
74     			if (req_state == 0 && c == '\r') {
75     				req_state++;
76     			} else if (req_state == 1 && c == '\n') {
77     				req_state++;
78     			} else if (req_state == 2 && c == '\r') {

pfalcon, 2018-06-07:
I thought that I marked such Coverity issues (on recv() error handling) as "Intended", because these samples are intended to be short, easy to read, and not to bloated with very detailed error handling.

closed 2018-06-08 by MaureenHelm