zephyrproject-rtos/zephyr #6095: [Coverity CID: 182776] Uninitialized variables in /tests/net/socket/udp/src/main.c

Shilpashree3lc, 2018-02-10:
Static code scan issues seen in File: /tests/net/socket/udp/src/main.c
 Category: Uninitialized variables
 Function: test_v6_sendto_recvfrom
 Component: Tests
 CID: 182776
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182776:  Uninitialized variables  (UNINIT)
/tests/net/socket/udp/src/main.c: 164 in test_v6_sendto_recvfrom()
158     
159     	prepare_sock_v6(CONFIG_NET_APP_MY_IPV6_ADDR,
160     			SERVER_PORT,
161     			&server_sock,
162     			&server_addr);
163     
>>>     CID 182776:  Uninitialized variables  (UNINIT)
>>>     Using uninitialized value "server_addr". Field 
>>> "server_addr.sin6_scope_id" is uninitialized when calling "bind". 
>>> [Note: The source code implementation of the function has been 
>>> overridden by a builtin model.]
164     	rv = bind(server_sock,
165     		  (struct sockaddr *)&server_addr,
166     		  sizeof(server_addr));
167     	zassert_equal(rv, 0, "bind failed");
168     
169     	test_sendto_recvfrom(client_sock,

pfalcon, 2018-05-29:
Fixed already.

closed 2018-05-29 by pfalcon