zephyrproject-rtos/zephyr #6098: [Coverity CID: 182773] Error handling issues in /samples/net/sockets/http_get/src/http_get.c

Shilpashree3lc, 2018-02-10:
Static code scan issues seen in File: /samples/net/sockets/http_get/src/http_get.c
 Category: Error handling issues
 Function: main
 Component: Samples
 CID: 182773
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182773:  Error handling issues  (CHECKED_RETURN)
/samples/net/sockets/http_get/src/http_get.c: 100 in main()
94     		}
95     
96     		response[len] = 0;
97     		printf("%s\n", response);
98     	}
99     
>>>     CID 182773:  Error handling issues  (CHECKED_RETURN)
>>>     Calling "close" without checking return value (as is done elsewhere 9 out of 11 times).

100     	close(sock);
101     
102     	return 0;

pfalcon, 2018-05-14:
Addressed (explicitly ignored) in https://github.com/zephyrproject-rtos/zephyr/pull/7517

closed 2018-05-14 by pfalcon