zephyrproject-rtos/zephyr #6100: [Coverity CID: 182771] Error handling issues in /samples/net/sockets/dumb_http_server/src/socket_dumb_http.c

Shilpashree3lc, 2018-02-10:
Static code scan issues seen in File: /samples/net/sockets/dumb_http_server/src/socket_dumb_http.c
 Category: Error handling issues
 Function: main
 Component: Samples
 CID: 182771
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182771:  Error handling issues  (NEGATIVE_RETURNS)
/samples/net/sockets/dumb_http_server/src/socket_dumb_http.c: 49 in main()
43     
44     	serv = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
45     
46     	bind_addr.sin_family = AF_INET;
47     	bind_addr.sin_addr.s_addr = htonl(INADDR_ANY);
48     	bind_addr.sin_port = htons(8080);
>>>     CID 182771:  Error handling issues  (NEGATIVE_RETURNS)
>>>     "serv" is passed to a parameter that cannot be negative. [Note: 
>>> The source code implementation of the function has been overridden 
>>> by a builtin model.]

49     	bind(serv, (struct sockaddr *)&bind_addr, sizeof(bind_addr));
50     
51     	listen(serv, 5);
52     
53     	while (1) {
54     		struct sockaddr_in client_addr;

pfalcon, 2018-05-14:
Addressed by https://github.com/zephyrproject-rtos/zephyr/pull/7526

closed 2018-05-14 by pfalcon

nashif, 2018-05-15:
You should not close bugs before the PR fixing them was merged...

pfalcon, 2018-05-15:
> You should not close bugs before the PR fixing them was merged...

1. Github is unable to cope with multiple issues #'s in "Fixes:" line.
2. There're multiple Coverity "proxy" issues on github for real Coverity CID's, some are now outdated, and some not covered. Fixes often apply to multiple issues. Summing up, there's a complete mess, the only reasonable way to deal with which is to "mark off" processed issues.