zephyrproject-rtos/zephyr #6103: [Coverity CID: 182768] Error handling issues in /samples/net/sockets/dumb_http_server/src/socket_dumb_http.c

Shilpashree3lc, 2018-02-10:
Static code scan issues seen in File: /samples/net/sockets/dumb_http_server/src/socket_dumb_http.c
 Category: Error handling issues
 Function: main
 Component: Samples
 CID: 182768
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

pfalcon, 2018-02-11:
@nashif, @jukkar : I'd appreciate a consult here. Indeed, error handling for all samples in `samples/net/sockets/` is rather simplified. That is to say, only "non-deterministically failable" errors, in the context of Zephyr, are being checked for. These samples are intended to be simple and clear ways to show off functionality supported by Zephyr, not a teaching material for writing robust networked applications (which is completely different matter). Performing complete error checking would (likely) turn these samples into tangles mess, and the need to apply them across number of samples is also not too bright.

So, I'd lean towards reviewing in some obvious checking needs to be added, and just mark these cases as "intended" in Coverity. What do you think?

nashif, 2018-02-11:
thats why those issues are marked as low, if there is no logical issues and it is only about error handling for example as in this case, you can mark it as intended. coverity however was very good discovering real issues in the past, thats why we do not dismiss them initially for samples and tests.