zephyrproject-rtos/zephyr #6105: [Coverity CID: 182766] Uninitialized variables in /tests/net/socket/udp/src/main.c

Shilpashree3lc, 2018-02-10:
Static code scan issues seen in File: /tests/net/socket/udp/src/main.c
 Category: Uninitialized variables
 Function: test_v6_bind_sendto
 Component: Tests
 CID: 182766
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182766:  Uninitialized variables  (UNINIT)
/tests/net/socket/udp/src/main.c: 248 in test_v6_bind_sendto()
242     
243     	rv = bind(client_sock,
244     		  (struct sockaddr *)&client_addr,
245     		  sizeof(client_addr));
246     	zassert_equal(rv, 0, "bind failed");
247     
>>>     CID 182766:  Uninitialized variables  (UNINIT)
>>>     Using uninitialized value "server_addr". Field 
>>> "server_addr.sin6_scope_id" is uninitialized when calling "bind". 
>>> [Note: The source code implementation of the function has been 
>>> overridden by a builtin model.]

248     	rv = bind(server_sock,
249     		  (struct sockaddr *)&server_addr,
250     		  sizeof(server_addr));
251     	zassert_equal(rv, 0, "bind failed");
252     
253     	test_sendto_recvfrom(client_sock,

pfalcon, 2018-04-23:
This should be fixed by edc048e06f21fa4bb4cf81cd117d24b4acdf5166.

closed 2018-04-23 by pfalcon