zephyrproject-rtos/zephyr #6106: [Coverity CID: 182765] Error handling issues in /tests/net/socket/udp/src/main.c

Shilpashree3lc, 2018-02-10:
Static code scan issues seen in File: /tests/net/socket/udp/src/main.c
 Category: Error handling issues
 Function: test_send_recv_2_sock
 Component: Tests
 CID: 182765
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182765:    (NEGATIVE_RETURNS)
/tests/net/socket/udp/src/main.c: 280 in test_send_recv_2_sock()
274     	sock1 = socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP);
275     	sock2 = socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP);
276     
277     	bind_addr.sin_family = AF_INET;
278     	bind_addr.sin_addr.s_addr = htonl(INADDR_ANY);
279     	bind_addr.sin_port = htons(55555);
>>>     CID 182765:    (NEGATIVE_RETURNS)
>>>     "sock1" is passed to a parameter that cannot be negative. [Note: 
>>> The source code implementation of the function has been overridden 
>>> by a builtin model.]

280     	bind(sock1, (struct sockaddr *)&bind_addr, sizeof(bind_addr));
281     
282     	conn_addr.sin_family = AF_INET;
283     	conn_addr.sin_addr.s_addr = htonl(0xc0000201);
284     	conn_addr.sin_port = htons(55555);
285     	connect(sock2, (struct sockaddr *)&conn_addr, sizeof(conn_addr));
/tests/net/socket/udp/src/main.c: 285 in test_send_recv_2_sock()
279     	bind_addr.sin_port = htons(55555);
280     	bind(sock1, (struct sockaddr *)&bind_addr, sizeof(bind_addr));
281     
282     	conn_addr.sin_family = AF_INET;
283     	conn_addr.sin_addr.s_addr = htonl(0xc0000201);
284     	conn_addr.sin_port = htons(55555);
>>>     CID 182765:    (NEGATIVE_RETURNS)
>>>     "sock2" is passed to a parameter that cannot be negative. [Note: 
>>> The source code implementation of the function has been overridden 
>>> by a builtin model.]

285     	connect(sock2, (struct sockaddr *)&conn_addr, sizeof(conn_addr));
286     
287     	send(sock2, BUF_AND_SIZE(TEST_STR_SMALL), 0);
288     
289     	len = recv(sock1, buf, sizeof(buf), 0);
290     	zassert_equal(len, 4, "Invalid recv len");

pfalcon, 2018-04-23:
Closed via #7156.

closed 2018-04-23 by pfalcon