zephyrproject-rtos/zephyr #6109: [Coverity CID: 182762] Control flow issues in /subsys/storage/flash_map/flash_map.c

Shilpashree3lc, 2018-02-10:
Static code scan issues seen in File: /subsys/storage/flash_map/flash_map.c
 Category: Control flow issues
 Function: get_flash_dev_from_id
 Component: Other
 CID: 182762
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

nvlsianpu, 2018-02-12:
@nashif Actullay both flash_map error are somehow false-positive (IMHO). I mean that it is intentionally possible to skip loop execution, and no `if` statement is need.
 
1) Should I cover this by if statement or additinal return an_error at the end?
```*** CID 182762:  Control flow issues  (NO_EFFECT)
/subsys/storage/flash_map/flash_map.c: 82 in get_flash_dev_from_id()
80     static struct device *get_flash_dev_from_id(u8_t id)
81     {
>>>     CID 182762:  Control flow issues  (NO_EFFECT)
>>>     This less-than-zero comparison of an unsigned value is never true. "i < 0UL".
82     	for (int i = 0; i < ARRAY_SIZE(flash_drivers_map); i++) {
83     		if (flash_drivers_map[i].id == id) {
84     			return flash_dev[i];
85     		}
86     	}
87
89    	k_panic();
90    }
```

2) Should I cover this by if statement?
```
*** CID 182194:  Control flow issues  (NO_EFFECT)
/subsys/storage/flash_map/flash_map.c: 271 in flash_map_init()
265     }
266     
267     static int flash_map_init(struct device *dev)
268     {
269     	int i;
270     
>>>     CID 182194:  Control flow issues  (NO_EFFECT)
>>>     This less-than-zero comparison of an unsigned value is never true. "i < 0UL".
271     	for (i = 0; i < ARRAY_SIZE(flash_dev); i++) {
272     		flash_dev[i] = device_get_binding(flash_drivers_map[i].name);
273     	}
274     
275     	return 0;
276     }
```

carlescufi, 2018-03-06:
@nvlsianpu can you fix those by adding an if() statement as appropriate?

closed 2018-03-27 by nvlsianpu