zephyrproject-rtos/zephyr #6121: doc: unit tests documentation refers to not existing sample code

aescolar-ot, 2018-02-10:
http://docs.zephyrproject.org/subsystems/test/ztest.html#quick-start-unit-testing
Refers to samples/testing/unit/ which does not exist anymore:
bd989a79bae2791f7365325c16769442e26cab51

nashif, 2018-02-10:
in all fairness, bug should be assigned to @jhedberg  who removed that code :)

jhedberg, 2018-02-10:
Ah, there's still a left-over in the documentation. I removed the part that CI grumpy, but seems this part didn't cause any CI warnings.

jhedberg, 2018-02-10:
Is it really so that the net_buf test was the only available unit test to make references to? It'd have been nicer to refer to some other test than remove the references complete.

jhedberg, 2018-02-10:
Actually, this is exactly what I removed from the documentation. Perhaps we just need to wait for the online documentation to be regenerated?

dbkinder, 2018-02-10:
FYI, docs are generated nightly about 11PM GMT

aescolar-ot, 2018-02-10:
My bad => I close it

closed 2018-02-10 by aescolar-ot