zephyrproject-rtos/zephyr #6127: net: Semantics of CONFIG_NET_BUF_POOL_USAGE changed (incorrectly)

pfalcon, 2018-02-11:
As part of dd09cbc1c455ab1e067b53f46bee7b6d50689bbc, for unknown reason, `config NET_BUF_POOL_USAGE` was put in `if NET_BUF_LOG` section. That setting is not related to Zephyr's logging and can be used, and actually used, independently of it.

Note that this is a typical outcome of massive, non-reviewable patches. This risk was raised in https://github.com/zephyrproject-rtos/zephyr/pull/5721, and suggestion was made to split that patch into parts, each dealing with a specific change, and "optimizing" Kconfig is apparently not related to the stated purpose of dd09cbc1c455ab1, which is "Redesigned API with split data and meta-data".

Thanks.

jhedberg, 2018-02-11:
I've never used the NET_BUF_POOL_USAGE myself, added @jukkar and @tbursztyka for some guidance

jhedberg, 2018-02-11:
@pfalcon or perhaps you have a patch suggestion, since it sounds like you're using this feature and will be able to verify the fix?

pfalcon, 2018-02-11:
Yep, if you look at the changes for net/Kconfig, you'll see that most changed options had "depends on NET_BUF_LOG", while NET_BUF_POOL_USAGE had "depends on NET_BUF":

https://github.com/zephyrproject-rtos/zephyr/commit/dd09cbc1c455ab1e067b53f46bee7b6d50689bbc#diff-f00090112a60a217eeaa95bd57a067d9L61

The solution is thus to move `endif # NET_BUF_LOG` bracket above that option. I can file a patch.

pfalcon, 2018-02-11:
(Found by building MicroPython port.)

jhedberg, 2018-02-11:
> The solution is thus to move endif # NET_BUF_LOG bracket above that option. I can file a patch.

I already uploaded a PR to fix this. Thanks for the report and fix suggestion.

PR #6129 (2018-02-11): net: buf: Remove incorrect dependency for NET_BUF_POOL_USAGE
NET_BUF_POOL_USAGE was never supposed to depend on NET_BUF_LOG.

Fixes #6127
commit 7aeb1fa1 (2018-02-11): net: buf: Remove incorrect dependency for NET_BUF_POOL_USAGE

NET_BUF_POOL_USAGE was never supposed to depend on NET_BUF_LOG.

Fixes #6127
closed 2018-02-11 by jhedberg