zephyrproject-rtos/zephyr #6147: "ninja flash" cannot be used with DFU-capable applications

SebastianBoe, 2018-02-12:
Hi,

as a user, when I do

cd bootloader_build_dir
ninja flash

cd app_build_dir
ninja flash

The second flash command will erase the bootloader written during the first flash command.

This makes "ninja flash" useless for applications that use a bootloader.

"ninja flash" is a useful feature that should be preserved for DFU-able applications.

SebastianBoe, 2018-02-12:
Perhaps this can be resolved by NOT erasing the entire flash when "ninja flash" is executed, but I don't understand why we are erasing it in the first place so I can't comment on if this is the correct approach ...

SebastianBoe, 2018-02-12:
@mbolivar : FYI

mbolivar, 2018-02-12:
@SebastianBoe I've been out sick for a few days, but am working on these issues

mbolivar, 2018-02-12:
> Perhaps this can be resolved by NOT erasing the entire flash when "ninja flash" is executed, but I don't understand why we are erasing it in the first place so I can't comment on if this is the correct approach ...

The entire flash is not erased by the application build system on the targets I use (pyOCD backend, but I've tested dfu-util in the past as well).

Could this be an artifact of your runner? I assume you are using nrfjprog.

SebastianBoe, 2018-02-13:
@mbolivar : Yes, I believe this is nrfjprog-specific, I thought I saw a patch go in recently that modified the nrfjprog runner to do erase-all, but I can't see evidence of this in the git history.

If pyOCD does not do erase-all then I definitely think the nrfjprog runner should align to not doing erase-all.

Perhaps this would be as simple as removing this line:

`+            ['nrfjprog', '--eraseall', '-f', self.family, '--snr', board_snr],`

from nrfjprog.py

mbolivar, 2018-02-13:
> Perhaps this would be as simple as removing this line:

OK! I can send a patch to make that optional (the way it is with e.g. jlink.py) if nrfjprog will still erase the sectors needed for the image it's flashing -- can you confirm that's the case?

SebastianBoe, 2018-02-13:
Sorry, there was a miscommunication. In addition to removing "--erase-all", "--sectorerase" must be appended to the "--program" command. It will ensure just the necessary pages are erased.

mbolivar, 2018-06-04:
Reopening since the west change is not in zephyr's copy of west yet -- it's a new feature, so not eligible for merge now.

nashif, 2018-11-05:
@mbolivar  still an issue?