zephyrproject-rtos/zephyr #6152: Inter-applications flash_map exchange mechanism

nvlsianpu, 2018-02-12:
Objective is to provide mechanism mechanism for provide information about flash layout for the mcuboot and  the application. Thanks to this mcuboot and application should been able to recognize proper drivers for memory devices they use. Not all of memory devices used by the app must be available for the mcuboot and vice versa.
It appear that storage of flash_area_map in certain flash page can provide such capability.
This will replace or will be able to override the current mechanism of provide such information statically (defined in compile time).
It is expected that Inter-application flash_map might changed during device life time - for instance this will allow to make bootloders' slots boundaries dynamic.

Good example of implementation of what should be achieved is  [Mynewt manufacturer mete region](https://github.com/apache/mynewt-core/tree/master/sys/mfg).

nvlsianpu, 2018-02-12:
@carlescufi @Laczen @pizi-nordic @anangl

Laczen, 2018-02-12:
Use a standard definition for fa_id and fa_device_id, but this will limit possible changes during lifetime.

Include the definition for fa_id and/or fa_device_id in flash. 

I would standardize fa_id and include the definition of fa_device_id (link between flash area and hardware device) in flash.

mbolivar, 2018-02-12:
Why is this beneficial to building this into mcuboot? Why is this a Zephyr issue and not an MCUboot issue?

Laczen, 2018-02-12:
@mbolivar: I think it is both an mcuboot and Zephyr issue. Both mcuboot and zephyr app could use flash_map, but I think mcuboot only needs the information required to boot, while the application might need more. Suppose you have a device with internal and external flash and mcuboot only uses the internal. Now if you would change the external flash hardware would you need to change mcuboot?

Adding to my previous comment I would define a standard flash interface (e.g. FLASH_DEV) that is not included in the flash definition of fa_device_id. More specific fa_device_id=0 means the standard flash device. This device is also where the flash_map definition in flash is located.

mbolivar, 2018-02-12:
> Both mcuboot and zephyr app could use flash_map

I've disagreed with this elsewhere. The flash_map is a mynewt API. There's nothing wrong with it, but Zephyr has its own flash API, and adding this new one on the side is not good design IMO.

It makes sense for the Zephyr port of MCUboot itself to have a flash_map implementation, since the bootloader core relies on it, and I know that support for it was added to zephyr in order to ease integration of other things like the FCB, but I don't like the way we have two APIs in Zephyr right now.

mbolivar, 2018-02-12:
Removing myself from the assignees list since I don't agree with the direction

carlescufi, 2018-02-12:
@mbolivar I actually wasn't fully aware that flash_map was conflicting in scope with the existing Zephyr flash APIs. If that is indeed the case then I also disagree with having this duplication. @nvlsianpu what's your take on this?

nvlsianpu, 2018-02-13:
@mbolivar I see your point. Our intention to port flash_map was also to introduce the abstraction over flash_drivers which allow easily operating across multiple different memories. If you have better idea we are open - At last @Laczen  saw cons of this what is in zephy for now.

I fill we misunderstand each other - this issue is about providing a mechanism which allow exchange the flash layout between bootloader and application and maybe backup-application in feature. And flash layout might be variable in life-cycle of the target. It not nesesery requires flash_map module at all. Does it make sense for you?

Laczen, 2018-02-13:
@nvlsianpu , @mbolivar , I don't see any cons of what is in zephyr for now. As far as I understand flash_map was added for FCB and it was completely ported to allow other mynewt modules to be easily ported e.g. for manufacturer meta region.

Laczen, 2018-02-13:
Oops, I was a little too fast. I guess what @nvlsianpu means is that I saw cons in the present flash_map implementation in zephyr. This is correct, I don't like the way the device is selected based upon the fa_device_id.

nvlsianpu, 2018-02-13:
cc @utzig

mbolivar, 2018-02-13:
I feel a @pfalcon rant coming on :wink:, sorry in advance ...

OK, since people have asked (here and in private email) what my objections are:

> @mbolivar I see your point. Our intention to port flash_map was also to introduce the abstraction over flash_drivers which allow easily operating across multiple different memories. If you have better idea we are open - At last @Laczen saw cons of this what is in zephy for now.

My main issues with sticking flash_map.h into Zephyr unmodified:

- It overlaps with the existing flash APIs in unclear ways. Consider the following APIs:

| flash_map.h                 | flash.h                    |
|-----------------------------|----------------------------|
| flash_area_get_sectors      | flash\*page\*              |
| flash_area_read             | flash_read                 |
| flash_area_write            | flash_write                |
| flash_area_erase            | flash_erase                |
| flash_area_align            | flash_get_write_block_size |

I understand that flash_map.h can abstract "areas" over multiple flash devices, unlike a single flash driver, but it's doing a **lot** more than that right now, in ways that we already have APIs for.

- It doesn't play well with the driver model. Since it was copied from mynewt, struct flash_area has a u8_t fa_device_id. Behind the scenes, there's a get_flash_dev_from_id() which converts that into a "real" struct device*. The extra layer is a hint that the API was bolted on IMO.

- It's not providing a general mechanism for users; instead it's "just enough" to get to the next MCUboot feature AFAICT. Using a hard-coded table of devices in flash_map.c, which currently just supports the built-in flash device and a W25Qxx SPI flash, gets us to the next MCUboot feature, namely storing the update off-die.  But any users that want to extend it will have to touch core Zephyr files themselves (instead of, say, updating their DTS flash partitions in an overlay file). Not a good experience; we should be trying to think of general mechanisms instead of continuing to increase the coupling between MCUboot and Zephyr.

- Speaking of DTS partitions, flash_map duplicates functionality provided by that mechanism in Zephyr as well. As far as I can tell, there's nothing connecting the flash_area_xxx APIs with those, which is another hint that it's been bolted on.

- It copies in code that we don't understand, and which we don't have clear use cases for, for reasons we're not sure about. I'm talking here about flash_area_open and flash_area_close, as discussed in #6026. This too is a sign that something is wrong.

> 
> I fill we misunderstand each other - this issue is about providing a mechanism which allow exchange the flash layout between bootloader and application and maybe backup-application in feature.

Why is that a useful feature for Zephyr to have? That was one of my first questions; I still don't understand why, sorry.

> And flash layout might be variable in life-cycle of the target.

> It not nesesery requires flash_map module at all. Does it make sense for you?

Sure, I'm in favor of slowly removing flash_map.h from Zephyr for more "native" APIs for the reasons I outlined above. I'm sorry I don't think I have time right now to help design those APIs. So that's why I removed myself from the assignees list -- basically to speak my piece and step out of the way.

mbolivar, 2018-02-13:
> My main issues with sticking flash_map.h into Zephyr unmodified:

BTW, those issues cover every declaration inside flash_map.h.

pfalcon, 2018-02-13:
> I feel a @pfalcon rant coming on 😉, sorry in advance ...

Oh, I don't even have bandwidth to follow the flash things, barely manage to follow net and general stuff... ;-)

mbolivar, 2018-02-13:
> Oh, I don't even have bandwidth to follow the flash things, barely manage to follow net and general stuff... ;-)

I meant from me, not you :)

nvlsianpu, 2018-02-21:
@mbolivar Thanks for elaboration - now I understand what you mean.

User case: Firmware update might changes flash layout of the device. This means: slots_images boundaries might change,  any user area boundaries might changes. So need a mechanism for transfer this information to the application (or even multiple application) and the bootloader. So fare flash area information are hardcoded in application and bootloader – idea is to make this information common for them all. Exchange mechanism should be optional of course.

I agree that its overlap with the driver - which is not good SW architecture.
Flash_map was a intermedium module - at the time when i introduce this to zephyr I didn't had enough time nor better conception of what should this looks like.

>I understand that flash_map.h can abstract "areas" over multiple flash devices, unlike a single flash driver.
No doubt: this is one of objective for having this at all. The module should support management over multiple memory devices - then I/O operations could be performed using flash API.
>It doesn't play well with the driver model. Since it was copied from mynewt, struct flash_area has a u8_t fa_device_id. Behind the scenes, there's a get_flash_dev_from_id() which converts that into a "real" struct device*. The extra layer is a hint that the API was bolted on IMO.
What is need is a way to identify proper drivers unambiguously; maybe the driver name string is better (as is natural to zephyr). Definitely is not a `structure device*` because it is compile time dependent. 

>It's not providing a general mechanism for users; instead it's "just enough" to get to the next MCUboot feature AFAICT. (...).

Think now moreless the same - DTS overlay should be the place where user can introduce layout. The reason for having a layout table I explained in <b>user case</b> above.

mbolivar, 2018-02-21:
Thanks for the response!

> User case: Firmware update might changes flash layout of the device.

Thanks for the concrete use case, which helps me understand.

That sounds like it will make MCUboot more complicated and thus more difficult to audit (dynamic partition maps -- will they be updated in a power-failure-safe way too? Will MCUboot need to parse a complex structure on flash to make that work? etc.). I don't see an open MCUboot issue for this additional feature; has it been discussed?

> So fare flash area information are hardcoded in application and bootloader – idea is to make this information common for them all. Exchange mechanism should be optional of course.

Regardless, I'm glad it will be optional -- the ability to bake a partition map permanently into the mcuboot binary (which should be immutable if it's the root of trust) is a security feature in my opinion.

> No doubt: this is one of objective for having this at all. The module should support management over multiple memory devices - then I/O operations could be performed using flash API.

+1 to this summary of what the API ought to do.

> What is need is a way to identify proper drivers unambiguously; maybe the driver name string is better (as is natural to zephyr). Definitely is not a structure device* because it is compile time dependent.

Seems workable.

> Think now moreless the same - DTS overlay should be the place where user can introduce layout. The reason for having a layout table I explained in user case above.

I think you're saying that a DT overlay should be used to describe the area map, which would get converted to a table by the build system. (Which is getting away from the flash_map.h API entirely at this point, especially if I/O is done with flash.h as you wrote above). Is that right?

That's interesting, though how to create "just" the binary blob for the table? Using the existing DT overlay support, there would be one table in every application image which uses that overlay. It seems like the requirement is to create a binary blob, which contains a flash area layout, and which can be shared by multiple application images.

nvlsianpu, 2018-03-13:
@utzig - just remind this issue to you.

utzig, 2018-03-13:
LOL, was just looking at this now, I was actually trying to find the cited zephyr flash map to see how it works...

galak, 2019-11-27:
Going to close this, re-open if you think this is still more work to be done.

closed 2019-11-27 by galak