zephyrproject-rtos/zephyr #6231: sample/bluetooth/eddystone: failed to connect with central device

tarunkum, 2018-02-16:
I am trying to flash the eddystone ble application on Micro-BIT device. It is failing to connect with central device.

Below are the steps which I followed on my machine :-

Steps followed on Linux machine :- 
sudo bluetoothctl 
[NEW] Controller 00:1A:7D:DA:71:13 localhost.localdomain [default]
[bluetooth]# power off
[bluetooth]# power on
[bluetooth]# agent KeyboardDisplay
[bluetooth]# default-agent on
[bluetooth]# scan on
[bluetooth]# scan on
Discovery started
[CHG] Device CA:92:9F:00:73:D6 Name: Zephyr Eddystone
[CHG] Device CA:92:9F:00:73:D6 Alias: Zephyr Eddystone
[bluetooth]# scan off
Discovery stopped
[bluetooth]# trust CA:92:9F:00:73:D6
[CHG] Device CA:92:9F:00:73:D6 Trusted: yes
Changing CA:92:9F:00:73:D6 trust succeeded
[bluetooth]# connect CA:92:9F:00:73:D6
Attempting to connect to CA:92:9F:00:73:D6
Failed to connect: org.bluez.Error.Failed
[bluetooth]# 

Console output ::-

***** BOOTING ZEPHYR OS v1.10.99 - BUILD: Feb 16 2018 04:52:21 *****
[bt] [INF] hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
[bt] [INF] hci_vs_init: HW Variant: nRF51x (0x0001)
[bt] [INF] hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 1.10 Build 99
[bt] [INF] show_dev_info: Identity: ca:92:9f:00:73:d6 (random)
[bt] [INF] show_dev_info: HCI: version 5.0 (0x09) revision 0x0000, manufacturer 0xffff
[bt] [INF] show_dev_info: LMP: version 5.0 (0x09) subver 0xffff
Bluetooth initialized
Configuration mode: waiting connections...
Switching to Beacon mode.

jhedberg, 2018-02-16:
Did you try connecting before Zephyr console shows "Switching to Beacon mode"? Beacon mode is a non-connectable state so connections would be expected to fail.

tarunkum, 2018-02-19:
@jhedberg

Thanks a lot. I understood to connect the device before Switching to Beacon mode.
Able to connect the device before it switches to beacon.

closed 2018-02-19 by jhedberg