zephyrproject-rtos/zephyr #6232: sample/bluetooth/central_hr: Run time HARD fault occurs

tarunkum, 2018-02-16:
I am trying to use two MICRO-BIT devices. In one device sample/Bluetooth/central_hr application is flashed and in another MICRO-BIT device sample/Bluetooth/peripheral application is flashed.

Once peripheral start advertising then central_hr application side seeing HARD fault occurs able to see on minicom.

Serial output for central_hr ::-

Bluetooth initialized
Scanning successfully started
[DEVICE]: fb:07:85:e8:9d:0a (random), AD evt type 0, AD data len 15, RSSI -63
[AD]: 9 data_len 10
[AD]: 1 data_len 1
[DEVICE]: fb:07:85:e8:9d:0a (random), AD evt type 4, AD data len 0, RSSI -63
[DEVICE]: 78:79:f2:f5:76:7d (random), AD evt type 0, AD data len 29, RSSI -44
[AD]: 1 data_len 1
[AD]: 3 data_len 6
[bt] [ERR] isr_rx_scan: assert: '!radio_is_ready()' failed
***** HARD FAULT *****
  Executing thread ID (thread): 0x200014c4
  Faulting instruction address:  0x108fe
Fatal fault in ISR! Spinning...

Serial output for peripheral  ::-

***** BOOTING ZEPHYR OS v1.10.99 - BUILD: Feb 16 2018 05:20:32 *****
[bt] [INF] hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
[bt] [INF] hci_vs_init: HW Variant: nRF51x (0x0001)
[bt] [INF] hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 1.10 Build 99
[bt] [INF] show_dev_info: Identity: ed:25:0e:36:81:ca (random)
[bt] [INF] show_dev_info: HCI: version 5.0 (0x09) revision 0x0000, manufacturer 0xffff
[bt] [INF] show_dev_info: LMP: version 5.0 (0x09) subver 0xffff
[bt] [WRN] bt_pub_key_gen: ECC HCI commands not available
[bt] [WRN] irk_init: Using temporary IRK
Bluetooth initialized
Advertising successfully started

jhedberg, 2018-02-16:
I think this is already fixed in the latest master branch, please re-test

cvinayak, 2018-02-16:
@tarunkum There was a regression that is fixed in https://github.com/zephyrproject-rtos/zephyr/commit/ae03e35d6cdd0642361917bc5dea5b6134de9da2

Please re-test and revert back.

tarunkum, 2018-02-19:
@jhedberg
 Yes Issue is fixed in latest branch. Thanks a lot.

closed 2018-02-19 by jhedberg