zephyrproject-rtos/zephyr #6233: samples/bluetooth/central: Run time HARD fault occurs

tarunkum, 2018-02-16:
I am trying to flash sample/Bluetooth/central_hr application on MICRO-BIT device.

I am finding HARD fault on serial console below ::

***** BOOTING ZEPHYR OS v1.10.99 - BUILD: Feb 16 2018 05:15:59 *****
Bluetooth initialized
Scanning successfully started
Device found: f7:77:01:62:d3:ec (random) (RSSI -85)
Device found: cc:ca:31:a1:b8:36 (random) (RSSI -89)
Device found: fb:07:85:e8:9d:0a (random) (RSSI -70)
***** HARD FAULT *****
  Executing thread ID (thread): 0x2000104c
  Faulting instruction address:  0xa626
Fatal fault in ISR! Spinning...

jhedberg, 2018-02-16:
I think this is already fixed in the latest master branch, please re-test

tarunkum, 2018-02-19:
@jhedberg 
Yes Issue is fixed in latest branch. Thanks a lot.

closed 2018-02-19 by jhedberg