zephyrproject-rtos/zephyr #6258: [Coverity CID: 182894] Error handling issues in /samples/net/nats/src/main.c

Shilpashree3lc, 2018-02-19:
Static code scan issues seen in File: /samples/net/nats/src/main.c
 Category: Error handling issues
 Function: write_led
 Component: Samples
 CID: 182894
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182894:  Error handling issues  (CHECKED_RETURN)
/samples/net/nats/src/main.c: 201 in write_led()
195     		fake_led = state;
196     	} else {
197     		gpio_pin_write(led0, LED_PIN, !state);
198     	}
199     
200     	pubstate = state ? "on" : "off";
>>>     CID 182894:  Error handling issues  (CHECKED_RETURN)
>>>     Calling "nats_publish(nats, "led0", 0U, msg->reply_to, 0U, pubstate, strlen(pubstate))" without checking return value. It wraps a library function that may fail and return an error code.
201     	nats_publish(nats, "led0", 0, msg->reply_to, 0,
202     		     pubstate, strlen(pubstate));
203     
204     	printk("*** Turning LED %s\n", pubstate);
205     }
206     

** CID 182893:    (OVERRUN)

satyacube, 2018-06-05:
In the process of submitting a PR for this issue. Can this issue be assigned to me?

PR #8176 (2018-06-06): samples: net: Check the return value of nats_publish
Check and handle the return value of nats_publish() in main.c. Also, apply
the changes suggested by the tools uncrustify and checkpatch.

Fixes #6258.
commit fd561dd5 (2018-06-06): samples: net: Check the return value of nats_publish

Check and handle the return value of nats_publish() in main.c

Fixes #6258.
closed 2018-06-06 by MaureenHelm