zephyrproject-rtos/zephyr #6259: [Coverity CID: 182892] Various in /tests/ztest/src/ztest.c

Shilpashree3lc, 2018-02-19:
Static code scan issues seen in File: /tests/ztest/src/ztest.c
 Category: Various
 Function: cleanup_test
 Component: Tests
 CID: 182892
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182892:    (DEADCODE)
/tests/ztest/src/ztest.c: 39 in cleanup_test()
33     	 * causes some problems.
34     	 */
35     	k_thread_abort(&ztest_thread);
36     #endif
37     
38     	if (!ret && mock_status == 1) {
>>>     CID 182892:    (DEADCODE)
>>>     Execution cannot reach this statement: "printf("Test %s failed: Unu...".
39     		PRINT("Test %s failed: Unused mock parameter values\n",
40     		      test->name);
41     		ret = TC_FAIL;
42     	} else if (!ret && mock_status == 2) {
43     		PRINT("Test %s failed: Unused mock return values\n",
44     		      test->name);
/tests/ztest/src/ztest.c: 43 in cleanup_test()
37     
38     	if (!ret && mock_status == 1) {
39     		PRINT("Test %s failed: Unused mock parameter values\n",
40     		      test->name);
41     		ret = TC_FAIL;
42     	} else if (!ret && mock_status == 2) {
>>>     CID 182892:    (DEADCODE)
>>>     Execution cannot reach this statement: "printf("Test %s failed: Unu...".
43     		PRINT("Test %s failed: Unused mock return values\n",
44     		      test->name);
45     		ret = TC_FAIL;
46     	}
47     
48     	return ret;

punitvara, 2018-05-29:
This seems to be false positive.

closed 2018-05-30 by punitvara