zephyrproject-rtos/zephyr #6260: [Coverity CID: 182890] Null pointer dereferences in /tests/net/net_pkt/src/main.c

Shilpashree3lc, 2018-02-19:
Static code scan issues seen in File: /tests/net/net_pkt/src/main.c
 Category: Null pointer dereferences
 Function: test_pkt_read_append
 Component: Tests
 CID: 182890
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182890:    (FORWARD_NULL)
/tests/net/net_pkt/src/main.c: 417 in test_pkt_read_append()
411     	zassert_true(!tfrag && fail_pos == 0xffff, "Invalid case NULL buffer");
412     
413     	/* Invalid: Skip more than a buffer length.*/
414     	tfrag = net_buf_frag_last(pkt->frags);
415     	tfrag = net_frag_skip(tfrag, tfrag->len - 1, &fail_pos, tfrag->len + 2);
416     	if (!(!tfrag && fail_pos == 0xffff)) {
>>>     CID 182890:    (FORWARD_NULL)
>>>     Dereferencing null pointer "tfrag".
417     		printk("Invalid case offset %d length to skip %d,"
418     		       "frag length %d\n",
419     		       tfrag->len - 1, tfrag->len + 2, tfrag->len);
420     		zassert_true(false, "Invalid offset");
421     	}
422     
/tests/net/net_pkt/src/main.c: 427 in test_pkt_read_append()
421     	}
422     
423     	/* Invalid offset */
424     	tfrag = net_buf_frag_last(pkt->frags);
425     	tfrag = net_frag_skip(tfrag, tfrag->len + 10, &fail_pos, 10);
426     	if (!(!tfrag && fail_pos == 0xffff)) {
>>>     CID 182890:    (FORWARD_NULL)
>>>     Dereferencing null pointer "tfrag".
427     		printk("Invalid case offset %d length to skip %d,"
428     		       "frag length %d\n",
429     		       tfrag->len + 10, 10, tfrag->len);
430     		zassert_true(false, "Invalid offset");
431     	}
432     
/tests/net/net_pkt/src/main.c: 443 in test_pkt_read_append()
437     	tfrag = pkt->frags;
438     	tfrag = tfrag->frags;
439     	off = tfrag->len;
440     	tfrag = net_frag_read(tfrag, off + 10, &tpos, 10, data);
441     	if (!tfrag ||
442     	    memcmp(sample_data + off + 10, data, 10)) {
>>>     CID 182890:    (FORWARD_NULL)
>>>     Dereferencing null pointer "tfrag".
443     		printk("Failed to read from offset %d, frag length %d "
444     		       "read length %d\n",
445     		       tfrag->len + 10, tfrag->len, 10);
446     		zassert_true(false, "Fail offset read");
447     	}
448

jukkar, 2018-03-26:
Pointer in question cannot be null.

closed 2018-03-26 by jukkar