zephyrproject-rtos/zephyr #6262: [Coverity CID: 182886] Error handling issues in /subsys/bluetooth/controller/hal/nrf5/ticker.c

Shilpashree3lc, 2018-02-19:
Static code scan issues seen in File: /subsys/bluetooth/controller/hal/nrf5/ticker.c
 Category: Error handling issues
 Function: hal_ticker_instance0_sched
 Component: Bluetooth
 CID: 182886
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182886:    (CHECKED_RETURN)
/subsys/bluetooth/controller/hal/nrf5/ticker.c: 67 in hal_ticker_instance0_sched()
61     			static memq_link_t link;
62     			static struct mayfly m = {0, 0, &link, NULL,
63     						  ticker_worker};
64     
65     			m.param = instance;
66     
>>>     CID 182886:    (CHECKED_RETURN)
>>>     Calling "mayfly_enqueue" without checking return value (as is done elsewhere 17 out of 19 times).
67     			mayfly_enqueue(TICKER_MAYFLY_CALL_ID_TRIGGER,
68     				       TICKER_MAYFLY_CALL_ID_WORKER,
69     				       chain,
70     				       &m);
71     		}
72     		break;
/subsys/bluetooth/controller/hal/nrf5/ticker.c: 113 in hal_ticker_instance0_sched()
107     			static memq_link_t link;
108     			static struct mayfly m = {0, 0, &link, NULL,
109     						  ticker_worker};
110     
111     			m.param = instance;
112     
>>>     CID 182886:    (CHECKED_RETURN)
>>>     Calling "mayfly_enqueue" without checking return value (as is done elsewhere 17 out of 19 times).
113     			mayfly_enqueue(TICKER_MAYFLY_CALL_ID_JOB,
114     				       TICKER_MAYFLY_CALL_ID_WORKER,
115     				       chain,
116     				       &m);
117     		}
118     		break;

carlescufi, 2018-03-06:
@cvinayak can this be fixed by Friday?

carlescufi, 2018-05-04:
Intentional, return value does not need to be checked. Commented in Coverity.

closed 2018-05-04 by carlescufi