zephyrproject-rtos/zephyr #6276: assert when running sys_kernel on disco_l475_iot1 (with asserts enabled)

nashif, 2018-02-20:
```
disco_l475_iot1           tests/benchmarks/sys_kernel/test                   building
MAKE: 'sanity_test_finished disco_l475_iot1/tests/benchmarks/sys_kernel/test'
DEVICE: ASSERTION FAIL [sys_clock_hw_cycles_per_tick <= (1 << 24)] @ /home/nashif/Work/github/zephyr/drivers/timer/cortex_m_systick.c:714:

```

nashif, 2018-02-20:
actually this looks like a common problem, not only on disco_l475_iot1, I was able to reproduce on frdm_k64f as well.

MaureenHelm, 2018-03-08:
Yes, this is a problem for any ARM platform running over 16 MHz. The test tries to eliminate timer interrupts by setting `CONFIG_SYS_CLOCK_TICKS_PER_SEC=1`, which overflows the systick timer. Can we run this test in tickless mode instead?

nashif, 2018-11-05:
can't reproduce anymore.

closed 2018-11-05 by nashif