zephyrproject-rtos/zephyr #6283: Build Fails on arm64 host

ric96, 2018-02-20:
```
linaro@linaro-developer:~/zephyr/samples/basic/blinky/build$ cmake -DBOARD=96b_carbon ..
CMake Deprecation Warning at /home/linaro/zephyr/cmake/app/boilerplate.cmake:38 (cmake_policy):
  The OLD behavior for policy CMP0000 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:1 (include)

-- Selected BOARD 96b_carbon
Zephyr version: 1.10.99
Parsing Kconfig tree in /home/linaro/zephyr/Kconfig
Using /home/linaro/zephyr/samples/basic/blinky/build/zephyr/.config as base
warning: HAS_DTS_SPI_DEVICE (defined at /home/linaro/zephyr/dts/Kconfig:38) has unsatisfied direct dependencies (HAS_DTS_SPI), but is currently being selected by the following symbols:
BOARD_96B_CARBON (defined at /home/linaro/zephyr/boards/arm/96b_carbon/Kconfig.board:8), with value y, direct dependencies SOC_STM32F401XE && <choice> (value: y), and select condition SOC_STM32F401XE && <choice> (value: y)
CMake Error at /home/linaro/zephyr/cmake/extensions.cmake:983 (message):
  No such file or directory: CMAKE_READELF: 'CMAKE_READELF-NOTFOUND'
Call Stack (most recent call first):
  /home/linaro/zephyr/cmake/compiler/gcc.cmake:14 (assert_exists)
  /home/linaro/zephyr/cmake/toolchain.cmake:38 (include)
  /home/linaro/zephyr/cmake/app/boilerplate.cmake:240 (include)
  CMakeLists.txt:1 (include)

-- Configuring incomplete, errors occurred!
linaro@linaro-developer:~/zephyr/samples/basic/blinky/build$ 
```

Debian Strech with updated cmake 3.9
the only issue with environment setup was that gcc-multilib wasn't installed.

andrewboie, 2018-02-20:
I don't think our build system has ever been tested (or supported) on anything other that x86 workstations.

MaureenHelm, 2018-03-23:
@nashif @galak Not sure who to assign this to, can you please take a look?

nashif, 2018-04-22:
this is not a bug, we also fail on many other architectures, right now x86 is recommended as the development environment, we will welcome any contributions that would make this work on arm64.