zephyrproject-rtos/zephyr #6291: userspace: support MMU-based memory virtualization

andrewboie, 2018-02-21:
Container issue; split into various MMU/VM related stories

- [ ] Virtualize address space, 2G/2G user/kernel VM split
- [ ] New APIs for supporting Zephyr "processes" which have their own VM, multiple Zephyr threads per process.
- [ ] Deprecate Memory Domains for MMU systems.
- [ ] ASLR
- [ ] Separately loaded position-independent application binaries loaded at runtime
- [ ] NuttX style demand paging
- [ ] x86: SMEP/SMAP

Deliver88, 2019-11-12:
Is there any activity to support container for Zephyr, e.g. approaches like BalenaEngine?

andrewboie, 2019-11-12:
> Hello, is there any activity to support container for Zephyr, e.g. approaches like BalenaEngine?

No this is not currently planned.

Deliver88, 2019-11-12:
> > Hello, is there any activity to support container for Zephyr, e.g. approaches like BalenaEngine?
> 
> No this is not currently planned.

Hi, thank you for the answer. Is there references how to setup/run a container on Zephyr? Thanks a lot

andrewboie, 2019-11-12:
> Is there references, ideas or approaches how to setup/run a container on Zephyr? Thanks a lot

There's no container support on Zephyr and nobody has indicated plans to investigate doing so.

Deliver88, 2019-11-13:
> > Is there references, ideas or approaches how to setup/run a container on Zephyr? Thanks a lot
> 
> There's no container support on Zephyr and nobody has indicated plans to investigate doing so.

thank you very much for answering! 

Is there an alternative / possibility in Zephyr to separate / hide applications from each other? container-like

andrewboie, 2019-11-13:
@Deliver88 what we have right now is documented here:

https://docs.zephyrproject.org/latest/reference/usermode/index.html

Provided virtual memory spaces for devices with an MMU and introducing the concept of Zephyr processes, is the next step and is being looked at this winter.

Deliver88, 2019-11-13:
> @Deliver88 what we have right now is documented here:
> 
> https://docs.zephyrproject.org/latest/reference/usermode/index.html
> 
> Provided virtual memory spaces for devices with an MMU and introducing the concept of Zephyr processes, is the next step and is being looked at this winter.

@andrewboie thanks a lot, it sounds good :)

nashif, 2021-04-20:
AFAIK most of this was done in the context of other issues/enhancements

closed 2021-04-20 by nashif