zephyrproject-rtos/zephyr #6309: Non-blocking BSD sockets not working as expected

jimmy-huang, 2018-02-21:
Symptom: When using the BSD sockets and running them as a server in non-blocking mode, the app will block on recv() until the client sends a message.

I am not sure if this is a bug or not, but at least the behavior is different from Linux implementation, from the linux man page, it states that if the socket is set to nonblocking mode, when there's no data coming from the client, then recv() should return the value -1and the external
variable errno is set to EAGAIN or EWOULDBLOCK.  However, running on Zephyr 1.10, the socket will block on recv() until the client sends a new message.

Note. there's no call to select() or poll(), and it appears that recv() only blocks after the client sends the 1st message, it didn't block initially when the client is connected.

Here's a sample code derived from the socket_echo sample, if you run this on Linux, it will continue to loop and print after each echoed string.  But when running on a frdm_k64f with Zephyr 1.10, it will block after client sends 1st message, .

Expected behavior (works on Linux):
This is normal - should not block on recv
This is normal - should not block on recv
This is normal - should not block on recv
..
..
.. (Continues)

// ----------------- Sample code to replicate ----------------------
#include <stdio.h>
#include <errno.h>

#ifndef __ZEPHYR__

#include <netinet/in.h>
#include <sys/socket.h>
#include <arpa/inet.h>
#include <unistd.h>
#include <fcntl.h>
#include <poll.h>

#else

#include <sys/fcntl.h>
#include <net/socket.h>
#include <kernel.h>
#include <net/net_app.h>

#endif

static void nonblock(int fd)
{
	int fl = fcntl(fd, F_GETFL, 0);
	fcntl(fd, F_SETFL, fl | O_NONBLOCK);
}

int main(void)
{
	int res;
	static int counter;
	int serv4;
	struct sockaddr_in bind_addr4 = {
		.sin_family = AF_INET,
		.sin_port = htons(4242),
		.sin_addr = {
			.s_addr = htonl(INADDR_ANY),
		},
	};

	serv4 = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
	res = bind(serv4, (struct sockaddr *)&bind_addr4, sizeof(bind_addr4));
	if (res == -1) {
		printf("Cannot bind IPv4, errno: %d\n", errno);
	}

	nonblock(serv4);
	listen(serv4, 5);

	printf("Listening on port 4242...\n");

	/* If server socket */
	int client = -1;
	struct sockaddr_storage client_addr;
	socklen_t client_addr_len = sizeof(client_addr);
	char addr_str[32];

	// loops until client connects
	while (1) {
		client = accept(serv4, (struct sockaddr *)&client_addr,
						&client_addr_len);
		if (client == -1) {
			if (errno != EAGAIN && errno != EWOULDBLOCK) {
				printf("Error on accept\n");
				return -1;
			}
#ifndef __ZEPHYR__
			sleep(1);
#else
			k_sleep(1000);
#endif
			continue;
		}
		void *addr = &((struct sockaddr_in *)&client_addr)->sin_addr;
		inet_ntop(client_addr.ss_family, addr, addr_str,
					sizeof(addr_str));
		printf("Connection #%d from %s fd=%d\n", counter++, addr_str,
				client);
		nonblock(client);
		break;
	}

	// loops until new message is received
	while (1) {
		char buf[128];
		int len = recv(client, buf, sizeof(buf), 0);
		if (len == -1) {
			if (errno != EAGAIN && errno != EWOULDBLOCK) {
				printf("Error on recv\n");
				close(serv4);
				return -1;
			}
			printf("This is normal - should not block on recv\n");
#ifndef __ZEPHYR__
			sleep(1);
#else
			k_sleep(1000);
#endif
			continue;
		}
		if (len == 0) {
			close(serv4);
			printf("Connection fd=%d closed\n", client);
			return 0;
		} else {
			send(client, buf, len, 0);
		}
	}
}

pfalcon, 2018-02-22:
@jimmy-huang : Ack, thanks, will look into it.

pfalcon, 2018-03-02:
@jimmy-huang : Sorry for the delay with looking into this (I'm working on a project which has a deadline next week). I can reproduce the issue with the code you supplied, thanks for it. Will try to look now if I find something wrong quickly...

jimmy-huang, 2018-03-03:
@pfalcon great, thanks for looking into it.

pfalcon, 2018-03-04:
@jimmy-huang: The specific issue with blocking recv() is fixed by https://github.com/zephyrproject-rtos/zephyr/pull/6445. While debugging the code, I found few more latent issues, some fixes included in the same PR, another documented in #6446, few more need additional analysis.

closed 2018-03-05 by carlescufi

jimmy-huang, 2018-03-05:
@pfalcon thanks, i tested it and it works well.

commit 5925112e (2018-03-05): net: sockets: Preserve user_data for any net_context send() or recv()

We use ctx->user_data to hold socket flags. As each call to
net_context_recv() and net_context_send() overwrites its previous
value, we explicitly must pass the current ctx value there.

Without this, non-blocking socket was turned into blocking after
e.g. switching from receiving to sending.

Fixes: #6309