zephyrproject-rtos/zephyr #6312: The shell sample does not working on k64f board

kkc0505, 2018-02-22:
When I test the shell on k64f, the following message is displayed after input command.

Zephyr Shell, Zephyr version: 1.10.99
Type 'help' for a list of available commands
shell> help
***** BUS FAULT *****
  Executing thread ID (thread): 0x200002d0
  Faulting instruction address:  0x224a
  Imprecise data bus error
Fatal fault in ISR! Spinning...

please help this issue.

thinks

MaureenHelm, 2018-02-22:
Thanks for reporting this. I'm still debugging, but after some git bisecting it looks like the introduction of mcumgr may be to blame

@carlescufi @ccollins476ad

ccollins476ad, 2018-02-22:
I don't believe I have access to a k64f, but I'll see if I can replicate it with other hardware.

MaureenHelm, 2018-02-22:
Fixed by #6313

closed 2018-02-22 by nashif

kkc0505, 2018-02-23:
Thinks for your comment.
I will check shell on k64f.

PR #6325 (2018-02-22): drivers: console: Fix crash in shell
Shell crashes when enter is pressed. This commit reverts relevant
changes from 14735116d1fe24713aa444e4e38e313b7f117c89 that cause
this issue.

Fixes #6322 
Fixes #6312