zephyrproject-rtos/zephyr #6319: Missing documentation for zephyr API to query kernel version: sys_kernel_version_get

pswarnak, 2018-02-22:
> sys_kernel_version_get() returns the kernel version of the current build.

It would be good to have this documented in website for general information to be used by product manufacturing teams.

In real products, kernel version is generally displayed as part of initial messages or crash log.
**This will help in debugging crash or non-reproducible scenarios.**

could not find related documentation http://docs.zephyrproject.org

dbkinder, 2018-02-22:
Best to report this as an issue on GitHub:  https://github.com/zephyrproject-rtos/zephyr/issues/new

BTW, looks like there are no doxygen comments in include/kernel_versoin.h where this function is defined ☹

-- david

nashif, 2018-02-23:
> Best to report this as an issue on GitHub:  https://github.com/zephyrproject-rtos/zephyr/issues/new

huh, and where is this? we are on github

dbkinder, 2018-02-23:
Doh. Sorry about that. I misread the address thinking this was posted to the mailing list.