zephyrproject-rtos/zephyr #6323: "SPI master port SPI_1 not found* when porting spi ethernet device enc28j60 on stm32_min_dev board

HorseMa, 2018-02-22:
I have enable SPI_1 in file stm32_min_dev.dts, but still error

HorseMa, 2018-02-22:
I add follow msg in file stm32_min_dev_defconfig, still error

# enable SPI
CONFIG_SPI_LEGACY_API=y
CONFIG_SPI=y
CONFIG_SPI_STM32=y
# CONFIG_SPI_STM32_INTERRUPT is not set
CONFIG_SPI_INIT_PRIORITY=70
CONFIG_SYS_LOG_SPI_LEVEL=0
CONFIG_SPI_CS_GPIO=y
# CONFIG_SPI_0 is not set
CONFIG_SPI_1=y
CONFIG_SPI_1_DEFAULT_CFG=0x80
CONFIG_SPI_1_DEFAULT_BAUD_RATE=500000
CONFIG_SPI_1_CS_GPIO_PORT="GPIOA"
CONFIG_SPI_1_CS_GPIO_PIN=4
# CONFIG_SPI_2 is not set
# CONFIG_SPI_3 is not set
# CONFIG_SPI_4 is not set
# CONFIG_SPI_5 is not set
# CONFIG_SPI_DW is not set
# CONFIG_SPI_DW_INTERRUPT_SINGLE_LINE is not set
# CONFIG_SPI_DW_INTERRUPT_SEPARATED_LINES is not set

erwango, 2018-02-22:
It's difficult to debug without seeing your changes, but possible issue is that you miss
` 	select HAS_DTS_SPI_DEVICE`
in boards/arm/stm32_min_dev/Kconfig.board

ydamigos, 2018-02-22:
Did you add the SPI pins in boards pinmux file? Commit c7a170b3731f5c46793aac2dfa6b9ae04533f1a3 contains information how it was done for olimexino_stm32

HorseMa, 2018-02-23:
I add the SPI pins in boards pinmux file， and my .config is flow, compile it , but there is nothing about spi in directory zephyr/CMakeFiles/zephyr.dir/drivers/

#
# Security Options
#
# CONFIG_STACK_CANARIES is not set
# CONFIG_USE_SWITCH is not set
# CONFIG_SMP is not set
CONFIG_MP_NUM_CPUS=1
# CONFIG_KERNEL_EVENT_LOGGER is not set
# CONFIG_SYS_POWER_MANAGEMENT is not set
CONFIG_HAS_DTS=y
# CONFIG_HAS_DTS_I2C is not set
CONFIG_HAS_DTS_SPI=y
CONFIG_HAS_DTS_SPI_DEVICE=y
# CONFIG_HAS_DTS_USB is not set

CONFIG_SPI_LEGACY_API=y
CONFIG_SPI=y
CONFIG_SPI_STM32=y
CONFIG_SPI_STM32_INTERRUPT=y
CONFIG_SPI_INIT_PRIORITY=70
CONFIG_SYS_LOG_SPI_LEVEL=4
CONFIG_SPI_CS_GPIO=y
# CONFIG_SPI_0 is not set
CONFIG_SPI_1=y
CONFIG_SPI_1_DEFAULT_CFG=0x80
CONFIG_SPI_1_DEFAULT_BAUD_RATE=500000
CONFIG_SPI_1_CS_GPIO_PORT="GPIOA"
CONFIG_SPI_1_CS_GPIO_PIN=4
# CONFIG_SPI_2 is not set
# CONFIG_SPI_3 is not set
# CONFIG_SPI_4 is not set
# CONFIG_SPI_5 is not set
# CONFIG_SPI_DW is not set

HorseMa, 2018-02-23:
SPI ethernet device enc28j60 depends on  CONFIG_SPI_LEGACY_API=y.
 I find if enable CONFIG_SPI_LEGACY_API the spi driver of stm32 would not to be compiled even if CONFIG_USE_STM32_LL_SPI=y,
why?

ydamigos, 2018-02-23:
STM32 SPI driver doesn't support SPI legacy API.
Could you try building PR https://github.com/zephyrproject-rtos/zephyr/pull/5921? It updates enc28j60 to the new SPI API.