zephyrproject-rtos/zephyr #6327: doc: GPIO_INT config option dependencies aren't clear

dbkinder, 2018-02-22:
As reported on the dev mailing list, there's a dependency between GPIO_INT_EDGE and GPIO_INT that's not clear. Can this dependency be added to the Kconfig file?

I don’t understand what the documentation says for the flags:

    GPIO_INT
       GPIO pin to trigger interrupt.

    GPIO_INT_EDGE
       Do Edge trigger.

So, what I’m concluding is that you need to specify GPIO_INT to **enable** interrupts on the pin and then **another flag** like GPIO_INT_EDGE to say **when in the signal processing** you want the interrupt to occur? If so... makes sense (now). I thought saying GPIO_INT_EDGE was enough to enable interrupts *and* to say they should occur on edge. My bad.

iha, 2018-02-25:
I agree. But I think it more intuitive to remove the flag to enable GPIO_INT in the documentation and leave it implicit:

```
/** @cond INTERNAL_HIDDEN */
/** GPIO pin to trigger interrupt. */
#define GPIO_INT                (1 << 1)
** @endcond */
 
/** GPIO pin trigger on level low or falling edge. */
#define GPIO_INT_ACTIVE_LOW     ((0 << 2) | GPIO_INT)
 
/** GPIO pin trigger on level high or rising edge. */
#define GPIO_INT_ACTIVE_HIGH    ((1 << 2) | GPIO_INT)
.
.
.
```

And using Kconfig to set this flag can be a bit annoying. I had to do something like this to set 16 GPIO ports for lack of API option.

mnkp, 2019-10-13:
Should be fixed by #16648 (not yet on master). Only the configuration flags which are supposed to be used by the application code have doxygen documentation. All of these compound flags explicitly include `GPIO_INT_ENABLE`.

carlescufi, 2020-02-11:
This has been addressed in the GPIO rework

closed 2020-02-11 by carlescufi