zephyrproject-rtos/zephyr #6333: How do I implement GPIO on the f429zi board?

kkc0505, 2018-02-23:
i am analyzing the source code related to gpio. that I saw the gpio_stm32 source.
I can not find the GPIO sample source code associated with the stm32 board.

How can I apply gpio to the f429zi board?

Please guide to me

ydamigos, 2018-02-23:
Hi,

did you check samples basic/button and basic/blinky?

erwango, 2018-02-23:
Can you precise which f429zi board you're using?

ydamigos, 2018-02-23:
Building basic/blinky (latest master) works on both nucleo_f429zi and stm32f429i_disc1. Which board do you use? Could you post the error?

kkc0505, 2018-02-23:
I'm using f429zi for the first time. ^^;;
blinky_led is not working. however blinky is working fine.

blinky_led and blinky are different samples.
samples/basic/blink_led/src/main.c:34:2: error: #error "Choose supported PWM driver"
 #error "Choose supported PWM driver"
  ^~~~~
samples/basic/blink_led/src/main.c: In function ‘main’:
samples/basic/blink_led/src/main.c:51:31: error: ‘PWM_DRIVER’ undeclared (first use in this function)
  pwm_dev = device_get_binding(PWM_DRIVER);
                               ^~~~~~~~~~
samples/basic/blink_led/src/main.c:51:31: note: each undeclared identifier is reported only once for each function it appears in
samples/basic/blink_led/src/main.c:58:33: error: ‘PWM_CHANNEL’ undeclared (first use in this function)
   if (pwm_pin_set_usec(pwm_dev, PWM_CHANNEL,
                                 ^~~~~~~~~~~
CMakeFiles/app.dir/build.make:302: recipe for target 'CMakeFiles/app.dir/src/main.c.obj' failed
make[2]: *** [CMakeFiles/app.dir/src/main.c.obj] Error 1
CMakeFiles/Makefile2:259: recipe for target 'CMakeFiles/app.dir/all' failed
make[1]: *** [CMakeFiles/app.dir/all] Error 2
Makefile:83: recipe for target 'all' failed

Thank you

ydamigos, 2018-02-23:
PWM support is not enabled on nucleo_f429zi and stm32f429i_disc1.
Check PR https://github.com/zephyrproject-rtos/zephyr/pull/6060 if you want to add PWM support on the board.

kkc0505, 2018-02-23:
Thanks everyone

closed 2018-02-23 by kkc0505