zephyrproject-rtos/zephyr #6338: Bluetooth: mesh: Node Identity Advertising issue

MariuszSkamra, 2018-02-23:
I read in the spec (7.2.2.2.3) that Advertising with Node Identity shall be started if Mesh Proxy Service is supported and provisioning is completed using PB-GATT. 
Now, Zephyr is starting Advertising with Node Identity even if provisioning was done using PB-ADV.
 I was about to report an issue to the PTS that they end the tests too early to see advertisements with Network ID, but now I have doubts. 
Could you please verify the implementation?

Affected TC:
MESH/NODE/CFG/GPXY/BV-02-C

When I changed Node Identity timeout, test is passing for me
CONFIG_BT_MESH_NODE_ID_TIMEOUT=5

BT_MESH_NODE_ID_TIMEOUT is set to 60 seconds by default.
In this test case (MESH/NODE/CFG/GPXY/BV-02-C), PTS is scanning about 10 seconds waiting for Network ID advertisements on two subnets only.

jhedberg, 2018-04-06:
I've uploaded a fix candidate for this in #6962 

@MariuszSkamra could you please test it? Also verify that it doesn't break any other provisioning or GATT proxy test cases.

MariuszSkamra, 2018-04-06:
@jhedberg Sure.
 MESH/NODE/CFG/GPXY/BV-02-C is passing now, I will check the rest test mesh cases

PR #6962 (2018-04-09): Bluetooth: Mesh: Fix Node Identity advertising with PB-ADV
The Node Identity advertising should only be automatically enabled
when provisioning happened over PB-GATT, but not when it happened over
PB-ADV. Move the enabling of Node Identity to the provisioning code,
where we know the bearer that was used (this information does not get
passed to the bt_mesh_provision function).

Fixes #6338
commit d6a549ce (2018-04-09): Bluetooth: Mesh: Fix Node Identity advertising with PB-ADV

The Node Identity advertising should only be automatically enabled
when provisioning happened over PB-GATT, but not when it happened over
PB-ADV. Move the enabling of Node Identity to the provisioning code,
where we know the bearer that was used (this information does not get
passed to the bt_mesh_provision function).

Fixes #6338
closed 2018-04-09 by jhedberg