zephyrproject-rtos/zephyr #6339: samples/drivers/gpio Sample doesn't work after SMP support was added

iha, 2018-02-23:
After this commit 837dd99a0e2b13bc4448b263918dd523a the gpio sample doesn't work. The sample get stuck.

andyross, 2018-02-23:
Yes, there are some regressions that snuck into the ESP-32 code when used with the asm2 layer.  Looking at this now.

If you're blocked, you can add CONFIG_XTENSA_ASM2=n to your kconfig as a workaround.

andyross, 2018-02-23:
Opened https://github.com/zephyrproject-rtos/zephyr/issues/6346 to track the low level details, which don't have anything to do with GPIO or this test in particular.  I have a handle on where I think the problem lies.

andyross, 2018-03-06:
@iha: The fix for the root cause is in master now.  Can you try this specific issue and verify it works for you now?

iha, 2018-03-06:
@andyross The behaviour is better now: the gpio input is triggered, but after several tests in succession it stops working, but the gpio output continues to run.

closed 2018-03-06 by andyross