zephyrproject-rtos/zephyr #6342: echo server: incorrect FCS and checksum in echo response

rmstoi, 2018-02-23:
Defensics test case 71191 ICMPv6.NDP.Router-advertisement-unsolicited.Router-Advertisement.IPv6-Packet.payload.protocol.icmpv6-message.ndp-router-advertisement.options.ndp-options.element

This test case fails with echo server in QEMU. echo server responds to echo request with malformed response. In the response Ethernet Frame Check Sequence and ICMPv6 checksum are incorrect.
[icmpv6-71191-replay.zip](https://github.com/zephyrproject-rtos/zephyr/files/1752359/icmpv6-71191-replay.zip)
[icmpv6-71191.zip](https://github.com/zephyrproject-rtos/zephyr/files/1752360/icmpv6-71191.zip)

laperie, 2018-03-05:
@rmstoi , can you please see if you can fix the bug as well?

rmstoi, 2018-03-05:
@laperie I will look into it

It seems also data in ICMPv6 echo response is malformed. Looks like a memory corruption.

PR #6536 (2018-03-12): net: ipv6: Drop RA with MTU out of valid range
This commit fixes incorrect Ethernet frame check sequence and ICMPv6
checksum caused by MTU set to zero in RA

Fixes #6342
commit 8109bdf8 (2018-03-12): net: ipv6: Drop RA with MTU out of valid range

This commit fixes incorrect Ethernet frame check sequence and ICMPv6
checksum caused by MTU set to zero in RA

Fixes #6342
closed 2018-03-12 by jukkar