zephyrproject-rtos/zephyr #6347: dhcpv4_client sample on spi_api_rework branch with board olimexino_stm32 is not work well

HorseMa, 2018-02-24:
it likes pend for ever。the log is below：

`[dev/enc28j60] [INF] eth_enc28j60_init: ENC28J60 Initialized
[dev/enc28j60] [DBG] eth_enc28j60_iface_init_0:
[net/dhcpv4] [DBG] dhcpv4_init: (0x20000834):
***** BOOTING ZEPHYR OS v1.10.99 - BUILD: Feb 24 2018 15:07:47 *****
[dhcpv4] [INF] main: Run dhcpv4 client
[net/dhcpv4] [DBG] net_dhcpv4_start: (0x20000834): state=init
[net/dhcpv4] [DBG] net_dhcpv4_start: (0x20000834): wait timeout=4s
[dev/enc28j60] [INF] eth_enc28j60_gpio_callback: ENC28J60 Interrupt
[dev/enc28j60] [DBG] eth_enc28j60_rx:
[dev/enc28j60] [DBG] eth_enc28j60_rx: Received packet of length 60
[net/ethernet] [DBG] ethernet_recv: (0x200005b4): Unknown hdr type 0x2600
[net/dhcpv4] [DBG] dhcpv4_timeout: (0x20000a40): state=init
[net/dhcpv4] [DBG] enter_selecting: (0x20000a40): enter state=selecting
[net/ethernet] [DBG] ethernet_send: (0x20000a40): src 00:04:A3:06:07:08 dst FF:FF:FF:FF:FF:FF type 0x800 len 114
[net/ethernet] [DBG] ethernet_send: (0x20000a40): src 00:04:A3:06:07:08 dst FF:FF:FF:FF:FF:FF type 0x800 len 114
[net/ethernet] [DBG] ethernet_send: (0x20000a40): src 00:04:A3:06:07:08 dst FF:FF:FF:FF:FF:FF type 0x800 len 49
[net/dhcpv4] [DBG] send_discover: (0x20000a40): send discover xid=0x2bb98e timeout=4s
[dev/enc28j60] [DBG] eth_net_tx: pkt 0x200016d4 (len 291)
[dev/enc28j60] [INF] eth_enc28j60_gpio_callback: ENC28J60 Interrupt
`

HorseMa, 2018-02-26:
ETH_ENC28J60_0_SPI_CS_PIN
ETH_ENC28J60_0_SLAVE

why define two times?

laperie, 2018-05-14:
@HorseMa , please try to reproduce the issue now after all SPI updates have been merged.  Is it still valid?

nashif, 2018-05-29:
closing, no response.

closed 2018-05-29 by nashif