zephyrproject-rtos/zephyr #6365: "dummy-flash" device tree bug

diegosueiro, 2018-02-25:
I'm currently working on porting Zephyr to i.MX7 SoC and I'm facing an error with cmake when it is generating the generated_dts_board.h.
I want to use only the sram and in my dts I removed the "zephyr,flash" property. I'm not sure if I'm doing this correctly. If I don't want a flash should I need to set it to a specific entry or just remove the property from the dts?

```
Error message:
Makefile:854: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
-- Selected BOARD colibri_imx7d_m4
Zephyr version: 1.10.99
Parsing Kconfig tree in <snip>/Kconfig
Using <snip>/tests/shell/build/colibri_imx7d_m4/zephyr/.config as base
-- Generating zephyr/include/generated/generated_dts_board.h
Traceback (most recent call last):
  File "<snip>/scripts/dts/extract_dts_includes.py", line 906, in <module>
    main()
  File "<snip>/scripts/dts/extract_dts_includes.py", line 896, in main
    insert_defs(chosen['zephyr,flash'], defs, load_defs, {})
KeyError: 'zephyr,flash'
CMake Error at <snip>/dts/dts.cmake:111 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  <snip>/cmake/app/boilerplate.cmake:227 (include)
  CMakeLists.txt:1 (include)
```

pfalcon, 2018-02-27:
@mbolivar : Maybe you can suggest something.

mbolivar, 2018-03-02:
@pfalcon there was a thread about this: https://lists.zephyrproject.org/pipermail/zephyr-devel/2018-February/008846.html

I only work in environments where flash is enabled, but it looked like @MaureenHelm had some concrete solutions for @diegosueiro to try that I don't see any additional information about in this issue report.

diegosueiro, 2018-03-09:
@mbolivar,

I updated the email thread, but basically:

There will be use cases where just one RAM will be available in the system for Zephyr and this RAM will have to accommodate both code and data. And to get this working I had to apply #6366 and omit the "zephyr,flash" property from dts.

galak, 2018-05-15:
Do you have an example that shows this issue?

diegosueiro, 2018-05-15:
@galak ,

I checked the current source of scripts/dts/extract_dts_includes.py and it changed a lot since I raised this issue.

Now, when I set CONFIG_XIP=n in boards/arm/colibri_imx7d_m4/colibri_imx7d_m4_defconfig file, the generated_dts_board.h generation goes through but the compilation fails:
```
real-ld: invalid origin for memory region FLASH
collect2: error: ld returned 1 exit status
zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:101: recipe for target 'zephyr/zephyr_prebuilt.elf' failed
make[2]: *** [zephyr/zephyr_prebuilt.elf] Error 1
CMakeFiles/Makefile2:576: recipe for target 'zephyr/CMakeFiles/zephyr_prebuilt.dir/all' failed
make[1]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```
Basically, the idea is to have both code and data segments in the same memory type. And when I raised this I was only be able to do that with the PR #6366.

diegosueiro, 2018-05-15:
@galak ,

If you want I can rebase the PR #6366 and apply this change to scripts/dts/extract_dts_includes.py.

diegosueiro, 2018-05-24:
@galak,

Can we consider the fix for the 1.12 release?

diegosueiro, 2018-06-23:
@galak ,

How are we going to fix this?

nashif, 2018-09-14:
@galak ?

galak, 2018-12-06:
@diegosueiro can you retest to see if this is still an issue.

diegosueiro, 2018-12-06:
> @diegosueiro can you retest to see if this is still an issue.

Sure. I'll be testing this in the weekend.

diegosueiro, 2018-12-08:
@galak,

I can confirm that at 1daf559 "dummy-flash" is working.

closed 2018-12-08 by diegosueiro