zephyrproject-rtos/zephyr #6370: I can't find adc name which is f429zi board

kkc0505, 2018-02-26:
Hi

Thanks very much for getting a lot of information here.
I am going to run the adc test on the f429zi board, but I do not know what the adc name is.

please guide to me

kkc0505, 2018-02-26:
i was test that ADC name is ADC1 and channel 5
- PA 5

it was not working, can't get device struct. which is NULL.

deadpoolcode1, 2019-09-05:
is there a future plan of adding ADC support? without that it's hard to use the STM32 boards for most use cases.

erwango, 2019-09-06:
@deadpoolcode1, ADC is now support on most STM32 devices

deadpoolcode1, 2019-09-06:
So zephyr documentation is not updated? ADC is supported for the STM32L?

erwango, 2019-09-06:
@deadpoolcode1, 

> So zephyr documentation is not updated? 

It should be, where are you looking at ?

> ADC is supported for the STM32L?

Supported on L0/L4. Effort for L1 should not be high