zephyrproject-rtos/zephyr #6384: Native (POSIX) zephyr.exe command line options not documented

dbkinder, 2018-02-26:
The resulting native POSIX application build creates a zephyr.exe program that can take command line options, but these options aren't documented.

Probably need to add this to http://docs.zephyrproject.org/getting_started/getting_started.html#running-a-sample-application-natively-posix-os where we can describe the --help option that lists what's supported:

    $ zephyr/zephyr.exe --help [-h] [--h] [--help] [-?] [-stop_at=<time>] [-seed=<r_seed>] [-testargs <arg>...] 

     [-h] [--h] [--help] [-?]     :Display this help
     [-stop_at=<time>]         :In simulated seconds, when to stop automatically
     [-seed=<r_seed>]          :Seed for the entropy device
     [-testargs <arg>...]         :Any argument that follows will be ignored by the top level, and made 
                                             available for possible tests

This also needs some updating to explain what the seed values can be:  

    A 32-bit integer seed value for the entropy device, such as 97229 (decimal),
   0x17BCD (hex), or 0275715 (octal)

aescolar-ot, 2018-05-10:
@dbkinder Currently the `--help` option is mentioned in http://docs.zephyrproject.org/boards/posix/native_posix/doc/board.html#how-to-use-it

Would you like to have this option also mentioned in http://docs.zephyrproject.org/getting_started/getting_started.html#running-a-sample-application-natively-posix-os?

Would you also like to have a list of all valid options? If so, in a new subsection of the native_posix board page?

The seed option is currently explained in the entropy device paragraph: http://docs.zephyrproject.org/boards/posix/native_posix/doc/board.html#peripherals

dbkinder, 2018-05-10:
@aescolar-ot  I'll submit a PR with a few more words and links from the getting started material to the native_posix board doc (no need to duplicate the info, but I'll add a reference.)  Stay tuned...