zephyrproject-rtos/zephyr #6388: entropy_native_posix doesn't follow "entropy" contract and is thus security risk

pfalcon, 2018-02-26:
An entropy driver for native_posix port, introduced in https://github.com/zephyrproject-rtos/zephyr/pull/6128, doesn't seem to generate real entropy. Its whole purpose seems to be the reproducible testing, but this aspect isn't emphasized enough (e.g. in its source filename, Kconfig description), and as such contradicts API contract for the entropy subsystem, which is, well, expected to generate entropy, not predictable values.

This may have implications up to security issues, for example, some cryptographic algorithms depend on true random nonce values, and may become insecure if predictable, or repeated values are used. If such an algorithm is used on the native posix port and communication available for eavesdropping, an attacker may recover security credentials used.

I obviously don't argue that there should not be a testing support for entropy drivers, but my recommendations would be:

1. Such a driver should be clearly named everywhere as "test", "mock", "fake", and "insecure".
2. Such a driver should print a user-visible notice about its insecurity on initialization.
3. I'd argue that mock testing shouldn't really be limited to native posix port, but is useful in general. (Thought arguably, availability of such a driver would add additional misuse risk - we can be sure that if there's a CONFIG_FAKE_MOCK_INSECURE_NON_ENTROPY setting, then there will be a vendor which will release a product with it enabled. However naming like that may help to draw a line between "vulnerability in Zephyr RTOS" and "vulnerability is specific product").

pfalcon, 2018-02-26:
This came during my work on https://github.com/zephyrproject-rtos/zephyr/issues/6086 . For it, I'm going to use check for CONFIG_ENTROPY_HAS_DRIVER, and if it's defined, assume that we have a true entropy, and are well set. If it's not defined, I'm going to print big fat scary warning, and use Zephyr's (P)RNG instead. Thus, usage of entropy_native_posix will bypass this big fat warning.

(All this comes as an outcome of earlier discussions with @d3zd3z. Note that I personally don't pledge for anything being resolved (I'd rather leave that to @d3zd3z and other security folks). At the same time, I feel it as my responsibility to raise this issue.)

aescolar-ot, 2018-02-27:
Adding further clarification in the documentation and Kconfig description sounds fair.

aescolar-ot, 2018-03-08:
Would it be a solution to add a new "UNSAFE_ENTROPY" config option like he proposed?. He could have a compile #error based on it.

d3zd3z, 2018-03-13:
I'm a little less worried about someone trying to enable mock entropy in order for predictable results (although I don't think this is actually useful, crypto libraries are generally unit tested with real entropy), than someone deciding to try and do bringup on a device that doesn't even have an entropy source. That's really the case where it seems likely for a product to get out with predictable entropy.

And, BTW, with predictable entropy, depending on the configuration, the work of attacking a particular stream either drops significantly, or drops very close to zero.

pfalcon, 2018-03-21:
> I would not consider a warning sufficient. It should fail to compile without a proper entropy source. I realize this is inconvenient, but the security risk is just too great to allow easy builds in insecure configurations.

@d3zd3z, the reason I submit this and similar tickets is to show that there's, a ... well, gap, between what's written in security team's docs, and what developers do in the source code. I definitely don't think that it makes sense to ignore this gap, and we need to look on ways to close it. Being on developers' side, I don't that the way to close it is ~~to make the code pain to work with~~ to move everything in one direction.

Good way would be to find compromise - both more security-aware, and yet comfortable for beginners and developers. E.g. compromise in this case could be:

1. Security team would consider warning (and other measures, like option renaming) *sufficient*.
2. Developers consider such measures *required*.

And let's not forget that it's all a process and dynamic - but jumping from a code full of  development shortcuts to one which isn't possible to develop or debug without extraordinary efforts seems like a big jump.

pfalcon, 2018-03-21:
> I'm a little less worried about someone trying to enable mock entropy in order for predictable results (although I don't think this is actually useful, crypto libraries are generally unit tested [...]

Sorry, but those are connect in a obvious way (in psychology-of-software-development sense): someone who'll do a bringup on a device without an entropy source will just enable a mock entropy source. Then may be ship it in a ready product. What we can do is to ensure that when doing so, they willfully try to do just that, and it's not result of confusion or misunderstanding. Hence, the option should (can) be named CONFIG_MOCK_INSECURE_NON_ENTROPY_DONT_USE_THIS, and there should (can) be big fat warning on startup.

aescolar-ot, 2018-05-10:
Is there some particular proposal that you have for the native entropy driver?
Would the following changes be satisfactory?
* Rename to driver file fake_entropy_native_posix.c
* Rename the Kconfig option to FAKE_ENTROPY_NATIVE_POSIX
* Expand the Kconfig help, and the comment in the file header, to indicate it shall only be used for testing, never for real products, and that it produces consistently the same random sequences.

I could also add a new Kconfig option called UNSAFE_ENTROPY and have this driver set it when selected..

nashif, 2018-05-15:
@aescolar-ot @pfalcon  Can we please have a resolution here? Is this going to be fixed/resolved for 1.12?

aescolar-ot, 2018-05-15:
@nashif: If @pfalcon & @d3zd3z were to agree that the proposal above is what they want, I'd be happy to do a PR.

pfalcon, 2018-05-16:
As I mentioned above, the main reason for submitting this ticket was to draw security team's attention to such issues - IMHO, we should have a general policy of dealing with such issues ("settings/code which *can* be a security issue in specific circumstances"), because individual developers' choices are unlikely to be consistent (and will range from doing nothing at all).

> Would the following changes be satisfactory?
> Rename to driver file fake_entropy_native_posix.c
> Rename the Kconfig option to FAKE_ENTROPY_NATIVE_POSIX
> Expand the Kconfig help, and the comment in the [...]

I personally would find these changes to be a good improvement, and would be happy to approve such a patch. But note that I also suggest to add an extra measure - print a warning of non-secure entropy source being used on the application startup. (That again shows that different developers have different choices, and a need for some policy and consistency.)

PR #7592 (2018-05-17): native: entropy: warn of security risk
Be more obvious about the entropy_native_posix driver
being only a test utility which does not generate real
entropy.

Fixes: #6388
commit 29ed87c9 (2018-05-17): native: entropy: warn of security risk

Be more obvious about the entropy_native_posix driver
being only a test utility which does not generate real
entropy.

Fixes: #6388
closed 2018-05-17 by nashif